openthread/openthread #2816: assert in otError MleRouter::HandleAdvertisement

NickNamerr, 2018-06-20:
Programm stuck on efr32.
 leader = mRouterTable.GetLeader();
                    assert(leader != NULL);
     0001 6094   B900       CBNZ         R0, 0x00016098             ; <_ZN2ot3Mle9MleRouter19HandleAdvertisementERKNS_7MessageERKNS_3Ip611MessageInfoE>+0x3BC
     0001 6096   E7FE       B            0x00016096                 ; <_ZN2ot3Mle9MleRouter19HandleAdvertisementERKNS_7MessageERKNS_3Ip611MessageInfoE>+0x3BA

NickNamerr, 2018-06-21:
I forked master branch at commit id 94bc7b14f30be7071ae3e1a871469d779042a30b.
Used Ozone from Segger as debugger.

jwhui, 2018-06-21:
I believe this issue was addressed with #2759.  Try moving to a more recent commit and see if the issue remains.

NickNamerr, 2018-06-29:
So far I have not faced issue again. Tks.