openthread/openthread #2842: "txpower 3" reset efr32mg13p732 ftd

NickNamerr, 2018-06-29:
Hello,

I could not set "txpower" for efr32mg13p732(supports up to 19dBm) from cli.
Code of "efr32/radio.c":

**_status = RAIL_SetTxPowerDbm(sRailHandle, ((RAIL_TxPower_t)OPENTHREAD_CONFIG_DEFAULT_TRANSMIT_POWER) * 10);_**

seems does not work, "txpower" in cli retruns "0" , "txpower 1" sets it to 1, "txpower 2" sets it to 1, "txpower 3" calls soft reset of the board.

jwhui, 2018-06-29:
I haven't tried to reproduce the issue yet, but I'm not able to notice any issues by reviewing the code.  Everything seems inline with the RAIL API documentation.

```
```

tomas-c, 2018-06-29:
From my experience RAIL_SetTxPowerDbm tries to set the closest value possible on the particular board. That value depends on your hal-config.h. From where did you take your hal-config.h? 

I would recommend testing with a SiLabs CLI application, it's very possible you'll see the same behaviour in which case will need to contact SiLabs. I had a similar problem with MGM12P.

NickNamerr, 2018-07-02:
Thank you, will try to reach out SiLabs support.

jwhui, 2018-07-09:
@NickNamerr, do you have any updates on this issue?

NickNamerr, 2018-07-17:
File radio.c
status = RAIL_SetTxPowerDbm(sRailHandle, ((RAIL_TxPower_t)aPower * **_10_**));
i could not enter float number to set 3 dBm (0.3)

in "void efr32RadioInit(void)" :
status = RAIL_SetTxPowerDbm(sRailHandle, ((RAIL_TxPower_t)OPENTHREAD_CONFIG_DEFAULT_TRANSMIT_POWER) * 10);
but :
../../openthread/openthread/src/core/openthread-core-default-config.h:#ifndef OPENTHREAD_CONFIG_DEFAULT_TRANSMIT_POWER
../../openthread/openthread/src/core/openthread-core-default-config.h:#define OPENTHREAD_CONFIG_DEFAULT_TRANSMIT_POWER 0

so 0 * 10 = 0

NickNamerr, 2018-07-17:
I do not have any updates, the main reason why i started to use OT is weak SiLabs support.

NickNamerr, 2018-07-17:
device:
> extaddr
ca880afc823c5a3f
Done
> txpower
17 dBm

OTBR:
wpanctl scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 1 |       NO | "OpenThreadDemo"   | 0x1234 | 11 | 1111111122222222 | CA880AFC823C5A3F |  -59

device:
> txpower 0
Done

OTBR:
wpanctl scan
   | Joinable | NetworkName        | PAN ID | Ch | XPanID           | HWAddr           | RSSI
---+----------+--------------------+--------+----+------------------+------------------+------
 4 |       NO | "OpenThreadDemo"   | 0x1234 | 11 | 1111111122222222 | CA880AFC823C5A3F |  -77

this is RAIL 2.2 issue. resolved it:
cd /home/dev/openthread/third_party/silabs/gecko_sdk_suite
ln -snf /home/dev/SimplicityStudio_v4/developer/sdks/gecko_sdk_suite/v2.3 v2.2

closed 2018-07-17 by NickNamerr

jwhui, 2018-07-20:
@NickNamerr, thanks for validating that this is an issue in Gecko SDK v2.2 and that it is addressed in v2.3.  I have submitted #2909 to update to Gecko SDK v2.3.

PR #2909 (2018-07-24): [efr32] update to Gecko SDK v2.3
Resolves #2842.