openthread/openthread #2843: Thread Network splits into partitions

NickNamerr, 2018-06-29:
Dear sir/madam
I have 31 devices with ot-cli-ftd installed. Sometimes device detaches of the network, form it own, became the "leader" and never returns back.
That means in same time i have 5-6 "leaders" on same channel, networkname, masterkey, panid, extended panid.

Could you tell how to prevent it?

NickNamerr, 2018-06-29:
1. gb833373, efr32mg13p732
2. I have to do it. could you give me reference how to get the log from device?

jwhui, 2018-06-29:
If you are using the CLI app, try:
1. Define `OPENTHREAD_CONFIG_LOG_OUTPUT` to be `OPENTHREAD_CONFIG_LOG_OUTPUT_APP` https://github.com/openthread/openthread/blob/c925b42941dc2bfc31161903e6754ff03a94489c/examples/platforms/efr32/openthread-core-efr32-config.h#L44
2. Add the `FULL_LOGS=1` to `make -f examples/Makefile-efr32` command. https://github.com/openthread/openthread/blob/eb4fcff15f92481a6193c0d52244f648f714b3cc/examples/common-switches.mk#L112

NickNamerr, 2018-06-29:
I caught  the log, but accidentally could not copy from putty terminal under UBUNTU, will send you screenshots of log over email.

tomas-c, 2018-06-29:
I think we've been mostly using the EFR32 driver for EFR32MG12P, yours is EFR32MG13P, perhaps that's why you are having issues.

Should be quite easy to get it to work though:

1) Get the correct hal-config.h from one of the demo Silabs projects for your radio.
2) Update all lines in <openthread>/examples/platforms/efr32/Makefile.am where it says EFR32MG12 or efr32mg12 (I see 4 such lines).
3) Update all lines <openthread>/examples/platforms/efr32/Makefile.platform.am where it says EFR32MG12 or efr32xg12 (I see 2 such lines).

NickNamerr, 2018-06-30:
The issue is not to get it work, but to figure out why thread network breaks in partitions?

jwhui, 2018-07-02:
@NickNamerr, thanks for providing the log.  The log does not show any MLE messages being received, which are required for partitions to merge.  What time period does this log capture cover?  Can you provide more logs across a longer time period?  We should see MLE Advertisements being sent by neighboring routers/leaders every 30 seconds.

NickNamerr, 2018-07-03:
@jwhui please check bigger log 
[log.log](https://github.com/openthread/openthread/files/2157339/log.log)

jwhui, 2018-07-03:
@NickNamerr, thanks for sharing the log.  It appears that there are no message buffers available to receive IPv6 datagrams.  Can you try using the `bufferinfo` CLI and sharing its output?  This should provide some additional insight into where the allocated buffers are being held.

NickNamerr, 2018-07-03:
This happens accidentally and very random. I need some time to run it and catch this case. But i will do it for sure.

NickNamerr, 2018-07-03:
Finally catch device  

>bufferinfo
total: 40
free: 0
6lo send: 22 40
6lo reas: 0 0
ip6: 0 0
mpl: 0 0
mle: 0 0
arp: 0 0
coap: 0 0
coap secure: 0 0
application coap: 0 0
Done
>state
leader
Done

after command 'reset' device came state became "child"

jwhui, 2018-07-03:
All of the buffers are queued in the mesh forwarder queue.  This behavior typically indicates an issue with the radio driver.  In particular, this tpyically occurs after a successful call to [otPlatRadioTransmit()](https://openthread.io/reference/group/radio-operation#otplatradiotransmit) but the radio driver does not call [otPlatRadioTxDone()](https://openthread.io/reference/group/radio-operation#otplatradiotxdone).

If you have a JTAG debugger, can inspect the radio driver state? For example:
- `sState`
- `sTransmitBusy`

jwhui, 2018-07-09:
Stopping and starting the Thread network interface puts the EFR32 radio back into a known state.  My guess is the radio driver is not calling `otPlatRadioTxDone()` after  a successful call to `otPlatRadioTransmit()`.

I would take a deeper look into the EFR32 radio driver.  The `sTransmitBusy` flag is generally reset as a part of [RAILCb_Generic()](https://github.com/openthread/openthread/blob/ed58a6775b732f629430c163a0749e0ea5d5c3f5/examples/platforms/efr32/radio.c#L711).  You may want to add some debugging code to see if the `sTransmitBusy` flag is getting reset properly.

NickNamerr, 2018-07-17:
Could it be a reason of subject:
	"C63FC1D598CF809A, RLOC16:0800, FrameErrRate:12.88%, MsgErrorRate:0.82%, AveRssi:-63, LastRssi:-64, "
	"DAEABD7F10574E09, RLOC16:0c00, FrameErrRate:8.10%, MsgErrorRate:0.74%, AveRssi:-60, LastRssi:-60, "
	"9A59E8FE4B807196, RLOC16:1c00, FrameErrRate:41.05%, MsgErrorRate:6.64%, AveRssi:-76, LastRssi:-76, "
	"023ADF571CB91F05, RLOC16:3c00, FrameErrRate:52.83%, MsgErrorRate:8.55%, AveRssi:-82, LastRssi:-83, "
	"FE0583C48C6F28C1, RLOC16:4800, FrameErrRate:44.46%, MsgErrorRate:5.66%, AveRssi:-73, LastRssi:-74, "
	"2A2F37163B2FB0D0, RLOC16:4c00, FrameErrRate:25.66%, MsgErrorRate:4.49%, AveRssi:-67, LastRssi:-66, "
	"FE83D2DFD1D2F0D5, RLOC16:5000, FrameErrRate:24.53%, MsgErrorRate:1.20%, AveRssi:-70, LastRssi:-70, "
	"3623693ECDB86209, RLOC16:5800, FrameErrRate:7.36%, MsgErrorRate:0.16%, AveRssi:-59, LastRssi:-60, "
	"267D06291718FA02, RLOC16:7000, FrameErrRate:36.77%, MsgErrorRate:2.66%, AveRssi:-77, LastRssi:-79, "
	"1A9045D3C648185E, RLOC16:7800, FrameErrRate:26.03%, MsgErrorRate:1.80%, AveRssi:-69, LastRssi:-71, "
	"DE8AE59797660469, RLOC16:9400, FrameErrRate:4.17%, MsgErrorRate:0.10%, AveRssi:-59, LastRssi:-60, "
	"7A13A4B31E2D3DD0, RLOC16:9800, FrameErrRate:46.07%, MsgErrorRate:5.59%, AveRssi:-80, LastRssi:-82, "
	"BE281972F19FE684, RLOC16:a800, FrameErrRate:66.66%, MsgErrorRate:20.45%, AveRssi:-85, LastRssi:-85, "
	"CAC3F311C2BDD6A2, RLOC16:bc00, FrameErrRate:10.32%, MsgErrorRate:0.10%, AveRssi:-70, LastRssi:-71, "
	"CA880AFC823C5A3F, RLOC16:e400, FrameErrRate:37.75%, MsgErrorRate:6.46%, AveRssi:-78, LastRssi:-88, "

FrameErrRate is to high.