openthread/openthread #2857: [nrf52840] Unable to write long bytes to uart

bukepo, 2018-07-04:
For CLI, pasting `dataset mgmtsetcommand pending pendingtimestamp 40 activetimestamp 80 delaytimer 60000 channel 16 panid 45006`, results in
```
> USBSmp 40 activetimestaUSBS
```
Using system call `write()` results in the same.

KaSroka, 2018-07-05:
What OS + terminal are you using? I have tried master with USB=0 on linux (ubuntu) with screen and minicom - everything is ok.

bukepo, 2018-07-05:
Did you paste or typed?

KaSroka, 2018-07-05:
As in your description - pasted.

KaSroka, 2018-07-05:
Yes, I've tested on PCA10056 0.11.0 using built-in JLink (JLink firmware from Jan 12 2018)

bukepo, 2018-07-06:
I'm testing on `PCA10056 V0.9.2 2017.26`, so this could be a hardware issue?

KaSroka, 2018-07-06:
I will test it on 0.9.2 2017.27 that I have. Can you try upgrading onboard JLink firmware so we could confirm that it's not an JLink issue?

KaSroka, 2018-07-06:
Quick test with minicom shows that it's working - but on this board I have same JLink firmware (I think it's from version 6.20i of their software)

bukepo, 2018-07-06:
@KaSroka How to upgrade the firmware? Do you have a guide?

KaSroka, 2018-07-06:
I don't know how to do this on Linux but on windows you can use J-Link Configurator (part of J-Link software for windows)

bukepo, 2018-07-09:
Hi @KaSroka , I updated J-Link firmware, the errors still exists. I verified screen on macOS and Linux. I verified using `write()` on Linux. All fails with the same error.
Here's the firmware info. As you see, it's the latest.
```
Disconnecting from J-Link...O.K.
Connecting to J-Link via USB...Updating firmware:  J-Link OB-SAM3U128-V2-NordicSemi compiled Jul  4 2018 16:22:02
Replacing firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled JUL  4 2018 16:22:02
Waiting for new firmware to boot
New firmware booted successfully
O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul  4 2018 16:22:02
Hardware version: V1.00
S/N: 683936325
VTref=3.300V
```

KaSroka, 2018-07-09:
Can you provide me the code using `write()` on Linux so I could test it on my side? I think this would allow me to make exactly same steps as you are making.

LuDuda, 2018-07-14:
Hello @bukepo,

Since we cannot reproduce this problem locally, and in my opinion problem has to be related with on-board J-Link and not with nRF52840 directly, i would recommend three things:

1) Run JLinkCommander or JLinkExe (under Linux), connect to the board and write: `msddisable` (it disables mass storage support)
2) Change JLink version to 5.12j (this is the most reliable JLink firmware that I know)
3) Check another dev-kit

And as @KaSroka wrote, if you can provide us the code, it would be easier to check what is going on.

bukepo, 2018-07-15:
Hi @LuDuda , I tried several dev boards, all behave the same. I also upgraded to the latest JLink firmware through JLinkExe and ran `msddisable`.
Here's what I did.
```
SEGGER J-Link Commander V6.32h (Compiled Jul  5 2018 18:13:32)
DLL version V6.32h, compiled Jul  5 2018 18:13:22

Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul  4 2018 16:22:02
Hardware version: V1.00
S/N: 683346769
VTref=3.300V

Type "connect" to establish a target connection, '?' for help
J-Link>usb
Disconnecting from J-Link...O.K.
Connecting to J-Link via USB...O.K.
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul  4 2018 16:22:02
Hardware version: V1.00
S/N: 683346769
VTref=3.300V
J-Link>msddisable
Probe configured successfully.
J-Link>exit
```
Then, I ran the test script
```python
import serial
import time

dev = serial.Serial('/dev/tty.usbmodem1A111', 115200, timeout=1)
dev.write(b'\r\n')
dev.read(1024)
dev.write(b'panid 0xface\r\n')
dev.flush()
time.sleep(1)
print(dev.read(1024))
dev.write(b'ifconfig up\r\n')
dev.flush()
time.sleep(1)
print(dev.read(1024))
dev.write(b'thread start\r\n')
dev.flush()
time.sleep(1)
print(dev.read(1024))
dev.write(b'dataset mgmtsetcommand pending pendingtimestamp 40 activetimestamp 80 delaytimer 60000 channel 16 panid 45006\r\n')
dev.flush()
print(dev.read(1024))
```
got the result,
```
b'panid 0xface\r\nDone\r\n> \r\n> '
b'ifconfig up\r\nDone\r\n> \r\n> '
b'thread start\r\nDone\r\n> \r\n> '
b'USBS~\x13\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 40 activetimestaUSBS~\x13\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```
As you see, there's some characters like **USBS** as I described in the issue description.

LuDuda, 2018-07-15:
Thanks @bukepo for trying out and sharing the test script. I will back to you ASAP.

LuDuda, 2018-07-17:
Hello @bukepo i was able to reproduce the issue on one of our Linux machine.

**Background**:

It seems that J-Link emulator on-board implements 6 USB endpoints (EPs) that can be used. 
Unfortunately, only 4 of them support USB packet sizes of up to 512 bytes. The EPs used for VCOM are the ones that do not support 512 byte packet sizes but only 64 bytes.

Strangly this seems not to be a problem on Windows, but only on Linux and probably on MAC.

More info here: [https://wiki.segger.com/J-Link-OB_SAM3U](https://wiki.segger.com/J-Link-OB_SAM3U)

**Solution**:

As i mentioned previously it helps if you disable MSD drag & drop feature and then **power cycle the board**.

**Logs**:

With MSD Enabled:

```
b'panid 0xface\r\nDone\r\n> \r\n> '
b'ifconfig up\r\nDone\r\n> \r\n> '
b'thread start\r\nDone\r\n> \r\n> '
b'USBSO\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 40 activetimestaUSBSO\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

With MSD Disabled (tested on J-link 5.12j and 6.20i and on dev-kits 0.9.2 and 0.10.0):

```
b'panid 0xface\r\nDone\r\n> \r\n> '
b'ifconfig up\r\nDone\r\n> \r\n> '
b'thread start\r\nDone\r\n> \r\n> '
b'dataset mgmtsetcommand pending pendingtimestamp 40 activetimestamp 80 delaytimer 60000 channel 16 panid 45006\r\nDone\r\n> \r\n> '
```

I modified your Python script a little bit to include larger delay after `thread start` command is called so that node may chance to become Child or Leader.

I guess alternative solution if you would like to use MSD feature is to overwrite **write** command to fragment the input into several 64-bytes long chunks, separated by small delay. Or obviously use UART pins directly without J-Link.

LuDuda, 2018-08-07:
Hello @bukepo, have you verified my suggestions? Is this still an issue?

jwhui, 2018-08-28:
@bukepo, any updates?

bukepo, 2018-08-29:
@LuDuda  After disabling MSD and flashing the latest firmware, it works! Thank you very much.

closed 2018-08-29 by bukepo