openthread/openthread #2896: [nrf52840] cc310 is not initialized in some cases

chshu, 2018-07-16:
CC310 is supposed to be initialized during platform init, but I noticed an issue that CC310 will not be initialized if OT dtls related features are not enabled (commissioning, joiner, border agent).

The way to reproduce:

1. Build with: ```make -f examples/Makefile-nrf52840 USB=1```
[cc310_mbedtls.c](https://github.com/openthread/openthread/blob/master/third_party/NordicSemiconductor/libraries/crypto/cc310_mbedtls.c) is **not** built in, the blank [nrf5CryptoInit()](https://github.com/openthread/openthread/blob/master/examples/platforms/nrf52840/misc.c#L45) is used instead (There seems no issue to do AES using CC310 without initializing it).

2. make -f examples/Makefile-nrf52840 COMMISSIONER=1 USB=1
[cc310_mbedtls.c](https://github.com/openthread/openthread/blob/master/third_party/NordicSemiconductor/libraries/crypto/cc310_mbedtls.c) is built in, and platform init will call into [nrf5CryptoInit()](https://github.com/openthread/openthread/blob/master/third_party/NordicSemiconductor/libraries/crypto/cc310_mbedtls.c#L44) to initialize CC310.

@rlubos Is it as expected?

rlubos, 2018-07-16:
@chshu No, it's not expected, CC310 should be initialized in both cases. The only case when it should not be built-in is when you specify DISABLE_CC310=1 during build explicitly. If it behaves as you say, there must've been a bug somewhere in build configuration.

chshu, 2018-07-16:
@rlubos I didn't specify DISABLE_CC310=1, could you please help to double confirm if the issue exists?

rlubos, 2018-07-16:
I'll try to reproduce it.

rlubos, 2018-07-16:
Ok, I was able to reproduce the issue. It looks that the problem we're facing is caused by some nasty "feature" of a `ld`. I've faced similar issue some time ago with different library.

So in general, by default linker searches for symbols only in those object files within library, that contain yet unresolved dependencies. Therefore, in case `weak` implementations are found by linker earlier than `strong` ones, `libmbedcrypto_a-cc310_mbedtls.o` will not even be processed anymore, as it does not contain other symbols.

There is an option to override this behavior - wrap the library inclusion within `--whole-archive` and `--no-whole-archive`, yet I'm not sure if it is possible to apply it easily with `autotools`.

So another option (more like a workaround) is to add other symbols to `libmbedcrypto_a-cc310_mbedtls.o`, that will prevent linker from dicarding it. Actually, there are some good candidates for that (`cc31_enable`/`cc31_disable`), so I think it would be the simplest solution. I'll propose a PR.

PR #2899 (2018-07-19): [nrf52840] prevent linker from dicarding strong nrf5CryptoInit implementation
Fixes #2896

Linker will discard libmbedcrypto_a-cc310_mbedtls.o when searching mbedTLS library, if it finds weak implementation of nrf5CryptoInit/Deinit before strong ones. Adding additional symbols to this file will prevent this behavior and make linker search libmbedcrypto_a-cc310_mbedtls.o even after weak implementations are found.
closed 2018-07-19 by jwhui