openthread/openthread #2904: Thread network keep "dead" ofMeshRoutes

NickNamerr, 2018-07-18:
0xd404  and 0x7806 are rloc16's of BR ncp that were before. Also remains configuration of GW.

Thread:NetworkData = [030F004020010DB88569B2B50103600A00]
Thread:NetworkDataVersion = 0xAC
Thread:OffMeshRoutes = [
	"2001:db8:8569:b2b5::/64    origin:ncp      stable:yes preference:medium  rloc:0xd404 next_hop_is_host:no "
	"2001:db8:8569:b2b5::/64    origin:ncp      stable:yes preference:medium  rloc:0x7806 next_hop_is_host:no "
	"2001:db8:8569:b2b5::/64    origin:user     stable:yes preference:medium  rloc:0x0000 next_hop_is_host:yes"
	"2001:db8:8569:b2b5::/64    origin:ncp      stable:yes preference:medium  rloc:0x600a next_hop_is_host:yes"
]
Thread:OnMeshPrefixes = [
	"2001:db8:8568:b2b5::   prefix_len:64   origin:ncp      stable:yes flags:0x33 [on-mesh:1 def-route:1 config:0 dhcp:0 slaac:1 pref:1 prio:med]"
]
Thread:Parent = "DAEABD7F10574E09, RLOC16:6000, Age:26, AveRssi:-61, LastRssi:-60, LQIn:3, LQOut:3"
Thread:PendingDataset = []
Thread:RLOC16 = 0x600A
Thread:RouterDowngradeThreshold = 0x17
Thread:RouterID = 0x18
Thread:RouterTable = [
	"9A59E8FE4B807196, RLOC16:0400, RouterId:1, NextHop:63, PathCost:0, LQIn:2, LQOut:2, Age:37, LinkEst:yes"
	"023ADF571CB91F05, RLOC16:0800, RouterId:2, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:62, LinkEst:yes"
	"1278E4AFCD902707, RLOC16:1800, RouterId:6, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:38, LinkEst:yes"
	"E6DE4D251653F63F, RLOC16:5000, RouterId:20, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:2, LinkEst:yes"
	"FE83D2DFD1D2F0D5, RLOC16:5c00, RouterId:23, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:40, LinkEst:yes"
	"0000000000000000, RLOC16:6000, RouterId:24, NextHop:63, PathCost:0, LQIn:0, LQOut:0, Age:91, LinkEst:no"
	"CA68FE2981E9647B, RLOC16:6800, RouterId:26, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:11, LinkEst:yes"
	"82F826170367DF49, RLOC16:8c00, RouterId:35, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:31, LinkEst:yes"
	"5ADBFBC6D309E9B6, RLOC16:ac00, RouterId:43, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:24, LinkEst:yes"
	"DE8AE59797660469, RLOC16:c800, RouterId:50, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:5, LinkEst:yes"
	"BE281972F19FE684, RLOC16:cc00, RouterId:51, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:2, LinkEst:yes"
	"8E3B262B73F035E3, RLOC16:d000, RouterId:52, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:61, LinkEst:yes"
	"C27C35C22540FAFF, RLOC16:d400, RouterId:53, NextHop:24, PathCost:1, LQIn:3, LQOut:3, Age:28, LinkEst:yes"
	"FE0583C48C6F28C1, RLOC16:dc00, RouterId:55, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:44, LinkEst:yes"
	"1A9045D3C648185E, RLOC16:f000, RouterId:60, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:57, LinkEst:yes"
	"461BABA7C4C5343F, RLOC16:f400, RouterId:61, NextHop:63, PathCost:0, LQIn:3, LQOut:3, Age:7, LinkEst:yes"
]
Thread:RouterUpgradeThreshold = 0x01
Thread:StableNetworkData = [030F004020010DB88569B2B50103FFFE00]

How to

librasungirl, 2018-07-18:
Generally it may take some time for the Leader to remove the stale border router information from network. If the border router is Child, the parent will be responsible for notifying the Leader that the border router not exist any more when timing out the child (default 240s).

Looking from your case,  
- for the 0xd404 
   the parent 0xd400 still live in the network, how long does 0xd404 has disappeared?
   If it doesn't exceed childtimeout(default 240s), it may make sense it is still in network data.

- for the 0x0x7806,  the parent 0x7800 is not live any more. There is one bug in existing implementation that when releasing one router, its children border router information not clean up.
Will propose one fix soon.

NickNamerr, 2018-07-18:
Eventually it cleaned, within couple hours.

NickNamerr, 2018-07-19:
I will do it as i get stable network.

jwhui, 2018-07-23:
Resolved by #2905.

closed 2018-07-23 by jwhui