openthread/openthread #2908: ot-cli-mtd FW stuck after ping BR mesh-local address

NickNamerr, 2018-07-20:
Pulled commit 1075e35327ec97e94404e3ecb0bb2381641423ee (origin/master, origin/HEAD).

After :
> ipaddr
2001:db8:8568:b2b5:2619:613d:ec45:d97
fd11:1111:1122:0:0:ff:fe00:4805
fe80:0:0:0:c888:afc:823c:5a3f
fd11:1111:1122:0:d5fd:c49b:ac26:c065
Done
> ping fd11:1111:1122:0:2060:fc4f:632b:4cd7
>
stuck in mesh_forwarder.cpp line 377:
 else if (netif.GetMle().IsMinimalEndDevice())
    {
        mMacDest.SetShort(netif.GetMle().GetNextHop(Mac::kShortAddrBroadcast));
    }
ot-cli-ftd works fine.

jwhui, 2018-07-20:
@NickNamerr, thank you for reporting this issue.  I was unable to reproduce this issue using posix emulation.

Can you provide all of the information necessary to reproduce and analyze?  For example:

- Hardware platform
- Build configuration
- Steps you performed in your test
- Debugger stack trace
- Was it working with an earlier commit?

NickNamerr, 2018-07-20:
Sorry, looks like i did not do "make clean; ./bootstrap" after "git fetch; git pull".

jwhui, 2018-07-20:
@NickNamerr, thanks for reporting back.

closed 2018-07-20 by jwhui