openthread/openthread #2913: Mesh hopping in thread

Bokenator, 2018-07-24:
I'm having a bit of problem reaching a device via the Mesh link multicast address `ff03::1`. My setup currently looks like the following:

```
A---B---C
```

The setup has 3 devices, A, B, and C. A is connected to B, which is connected to C, but A and C are out of range from each other. 

The problem I am having right now is that I am able to ping C from A by using C's link global address. But I am not able to ping C from A through `ff03::1`. If I ping `ff03::1` from A, only B responds. On the other hand, if I ping `ff03::1` from B, both A and C responds. 

My question is, would there be a way to multicast to all the router devices in a thread network, even routers that are not directly connected to the device issuing the multicast?

jwhui, 2018-07-24:
Can you provide more information on your setup?  Are the nodes running in SoC mode (e.g. with CLI app) or NCP mode (e.g. with wpantund on Linux)?

Bokenator, 2018-07-24:
@jwhui In the setup above, all three devices are running NCP with wpantund on Linux, and I am running the ping commands from the shell.

jwhui, 2018-07-24:
Linux, by default, sets the IPv6 Hop Limit value to 1 on multicast messages.  Check if your `ping6` command offers a "hop limit" or "ttl" option and set that to something larger than 1.

Bokenator, 2018-07-25:
```
pi@raspberrypi:/etc $ ping -h
Usage: ping [-aAbBdDfhLnOqrRUvV64] [-c count] [-i interval] [-I interface]
            [-m mark] [-M pmtudisc_option] [-l preload] [-p pattern] [-Q tos]
            [-s packetsize] [-S sndbuf] [-t ttl] [-T timestamp_option]
            [-w deadline] [-W timeout] [hop1 ...] destination
Usage: ping -6 [-aAbBdDfhLnOqrRUvV] [-c count] [-i interval] [-I interface]
             [-l preload] [-m mark] [-M pmtudisc_option]
             [-N nodeinfo_option] [-p pattern] [-Q tclass] [-s packetsize]
             [-S sndbuf] [-t ttl] [-T timestamp_option] [-w deadline]
             [-W timeout] destination
```

@jwhui Unfortunately in Raspberry Pi (running Raspbian) I'm unable to find the flag that sets the maximum number of hops allowed.

Bokenator, 2018-07-25:
```
pi@raspberrypi:/etc $ ping -v -I wpan0  ff03::1
ping: socket: Permission denied, attempting raw socket...
ping: socket: Permission denied, attempting raw socket...
PING ff03::1(ff03::1) from fdde:ad00:beef:0:24ae:741a:683d:d606 wpan0: 56 data bytes
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=1 ttl=64 time=0.189 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=1 ttl=64 time=11.9 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=2 ttl=64 time=0.138 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=2 ttl=64 time=5.08 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=3 ttl=64 time=0.147 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=3 ttl=64 time=5.79 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=4 ttl=64 time=0.147 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=4 ttl=64 time=15.5 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=5 ttl=64 time=0.136 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=5 ttl=64 time=11.7 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=6 ttl=64 time=0.141 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=6 ttl=64 time=7.21 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=7 ttl=64 time=0.136 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=7 ttl=64 time=9.69 ms (DUP!)
```

In this case I'm running the ping command from device A. I'm just issuing the ping command with the default ttl, and it looks like the default value is 64. All the output with (DUP!) are returned from device B. So far nothing is coming back from device C.

jwhui, 2018-07-25:
Did you try running the command with sudo?

Bokenator, 2018-07-25:
```
pi@raspberrypi:/etc $ sudo ping -v -I wpan0  ff03::1
ping: socket: Permission denied, attempting raw socket...
ping: socket: Permission denied, attempting raw socket...
PING ff03::1(ff03::1) from fdde:ad00:beef:0:24ae:741a:683d:d606 wpan0: 56 data bytes
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=1 ttl=64 time=0.143 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=1 ttl=64 time=10.5 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=2 ttl=64 time=0.141 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=2 ttl=64 time=93.3 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=3 ttl=64 time=0.133 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=3 ttl=64 time=40.0 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=4 ttl=64 time=0.133 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=4 ttl=64 time=18.2 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=5 ttl=64 time=0.134 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=5 ttl=64 time=16.2 ms (DUP!)
64 bytes from fdde:ad00:beef:0:24ae:741a:683d:d606: icmp_seq=6 ttl=64 time=0.133 ms
64 bytes from fdde:ad00:beef:0:f9f6:3bc:84d1:ada6: icmp_seq=6 ttl=64 time=19.4 ms (DUP!)
```

The result seems to be the same.

jwhui, 2018-07-25:
Can you use tcpdump to capture the packets going into wpan0 in a pcap file?

Bokenator, 2018-07-25:
```
pi@raspberrypi:/etc $ sudo tcpdump -i wpan0                                                                                                                                                                                                   
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wpan0, link-type RAW (Raw IP), capture size 262144 bytes
17:18:15.641307 IP6 fdde:ad00:beef:0:24ae:741a:683d:d606 > ff03::1: ICMP6, echo request, seq 1, length 64
17:18:15.887154 IP6 fdde:ad00:beef:0:f9f6:3bc:84d1:ada6 > fdde:ad00:beef:0:24ae:741a:683d:d606: ICMP6, echo reply, seq 1, length 64
17:18:16.642380 IP6 fdde:ad00:beef:0:24ae:741a:683d:d606 > ff03::1: ICMP6, echo request, seq 2, length 64
17:18:16.889680 IP6 fdde:ad00:beef:0:f9f6:3bc:84d1:ada6 > fdde:ad00:beef:0:24ae:741a:683d:d606: ICMP6, echo reply, seq 2, length 64
17:18:17.643859 IP6 fdde:ad00:beef:0:24ae:741a:683d:d606 > ff03::1: ICMP6, echo request, seq 3, length 64
17:18:17.886254 IP6 fdde:ad00:beef:0:f9f6:3bc:84d1:ada6 > fdde:ad00:beef:0:24ae:741a:683d:d606: ICMP6, echo reply, seq 3, length 64
17:18:18.645416 IP6 fdde:ad00:beef:0:24ae:741a:683d:d606 > ff03::1: ICMP6, echo request, seq 4, length 64
17:18:18.890807 IP6 fdde:ad00:beef:0:f9f6:3bc:84d1:ada6 > fdde:ad00:beef:0:24ae:741a:683d:d606: ICMP6, echo reply, seq 4, length 64
```

I just copy and pasted the output. Does that help?

jwhui, 2018-07-25:
Not quite.  The idea is to view the IPv6 Hop Limit value in the packet.  Can you try the `-vv` option?  If that doesn’t fully decode the IPv6 header, then capturing the pcap file would allow using wireshark to dissect the packets.

jwhui, 2018-07-25:
Yes, note that the IPv6 Hop Limit “hlim” value is set to 1.  That’s why the packet is only traversing a single hop.

Does ping6 exist on your RPi?  If so, can you try that?

Bokenator, 2018-07-25:
Yes, I do have ping6 on my RPi. However, I'm not sure which flag I should use to change the hlim parameter.

```
pi@raspberrypi:/etc $ ping6 -h
Usage: ping [-aAbBdDfhLnOqrRUvV64] [-c count] [-i interval] [-I interface]
            [-m mark] [-M pmtudisc_option] [-l preload] [-p pattern] [-Q tos]
            [-s packetsize] [-S sndbuf] [-t ttl] [-T timestamp_option]
            [-w deadline] [-W timeout] [hop1 ...] destination
Usage: ping -6 [-aAbBdDfhLnOqrRUvV] [-c count] [-i interval] [-I interface]
             [-l preload] [-m mark] [-M pmtudisc_option]
             [-N nodeinfo_option] [-p pattern] [-Q tclass] [-s packetsize]
             [-S sndbuf] [-t ttl] [-T timestamp_option] [-w deadline]
             [-W timeout] destination
```

jwhui, 2018-07-25:
In the examples you included above, none included the `-t` option.  Did you try the following?
```
sudo ping6 -I wpan0 -t 64 ff03::1
```

Bokenator, 2018-07-25:
@jwhui That worked! Thanks a lot!

closed 2018-07-25 by Bokenator