openthread/openthread #2921: Broken build after enabling OPENTHREAD_CONFIG_DISABLE_CSMA_CA_ON_LAST_ATTEMPT

tomas-c, 2018-07-25:
When compiling latest master (bb064cd074939cddb0d4020e772857584d147dfd) for EFR32 with OPENTHREAD_CONFIG_DISABLE_CSMA_CA_ON_LAST_ATTEMPT set to 1 I got the following error:

`
/home/tomas/Work/repos/openthread/examples/../src/core/mac/mac.cpp: In member function 'void ot::Mac::Mac::StartCsmaBackoff()':
/home/tomas/Work/repos/openthread/examples/../src/core/mac/mac.cpp:1032:15: error: 'sendFrame' was not declared in this scope
     else if ((sendFrame.GetMaxFrameRetries() > 0) && (sendFrame.GetMaxFrameRetries() <= mTransmitRetries))
               ^~~~~~~~~
/home/tomas/Work/repos/openthread/examples/../src/core/mac/mac.cpp:1032:15: note: suggested alternative: 'mTxFrame'
     else if ((sendFrame.GetMaxFrameRetries() > 0) && (sendFrame.GetMaxFrameRetries() <= mTransmitRetries))
               ^~~~~~~~~
               mTxFrame
/home/tomas/Work/repos/openthread/examples/../src/core/mac/mac.cpp: In member function 'void ot::Mac::Mac::BeginTransmit()':
/home/tomas/Work/repos/openthread/examples/../src/core/mac/mac.cpp:1148:5: error: 'else' without a previous 'if'
     else if ((sendFrame.GetMaxFrameRetries() > 0) && (sendFrame.GetMaxFrameRetries() <= mTransmitRetries))
     ^~~~
`

jwhui, 2018-07-25:
Thanks for raising this issue.  I'll take a look.

jwhui, 2018-07-25:
I submitted #2922.  Please review.

tomas-c, 2018-07-25:
Thanks! Fixes the compilation problem.

PR #2922 (2018-07-26): [mac] fix build failure with CONFIG_DISABLE_CSMA_CA_ON_LAST_ATTEMPT
Resolves #2921.
closed 2018-07-26 by jwhui