openthread/openthread #2931: gcc8 compiler error: class-memaccess

PlaidShirtPat, 2018-07-26:
I am trying to compile the posix example platform and getting the following error:
```bash
$ make -f examples/Makefile-posix 
  BUILD    x86_64-unknown-linux-gnu
make -j10 -C build/x86_64-unknown-linux-gnu --no-print-directory all
--- SNIP ---

/openthread/examples/../src/core/net/dhcp6_client.cpp: In member function ‘otError ot::Dhcp6::Dhcp6Client::Solicit(uint16_t)’:
/openthread/examples/../src/core/net/dhcp6_client.cpp:390:48: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘class ot::Ip6::MessageInfo’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&messageInfo, 0, sizeof(messageInfo));
                                                ^

/openthread/examples/../src/core/meshcop/dataset.cpp: In member function ‘otError ot::MeshCoP::Dataset::AppendMleDatasetTlv(ot::Message&) const’:
/openthread/examples/../src/core/meshcop/dataset.cpp:492:56: error: ‘void* memcpy(void*, const void*, size_t)’ copying an object of non-trivial type ‘class ot::MeshCoP::DelayTimerTlv’ from an array of ‘const class ot::MeshCoP::Tlv’ [-Werror=class-memaccess]
             memcpy(&delayTimer, cur, sizeof(delayTimer));
                                                        ^
In file included from /openthread/examples/../src/core/meshcop/dataset.hpp:42,
                 from /openthread/examples/../src/core/meshcop/dataset.cpp:37:
/openthread/examples/../src/core/meshcop/meshcop_tlvs.hpp:1234:7: note: ‘class ot::MeshCoP::DelayTimerTlv’ declared here
 class DelayTimerTlv : public Tlv
       ^~~~~~~~~~~~~
/openthread/examples/../src/core/net/dhcp6_server.cpp: In member function ‘otError ot::Dhcp6::Dhcp6Server::SendReply(otIp6Address&, uint8_t*, ot::Dhcp6::ClientIdentifier&, ot::Dhcp6::IaNa&)’:
/openthread/examples/../src/core/net/dhcp6_server.cpp:431:48: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘class ot::Ip6::MessageInfo’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&messageInfo, 0, sizeof(messageInfo));

```

This on Archlinux 4.17.6-1
gcc version: gcc (GCC) 8.1.1 20180531
commit: 98ed5ea2adf9a7e2641e6ac70fb0bc3c2c7fc74e

It seems this is a newish compiler error introduced in gcc8

jwhui, 2018-07-26:
Thanks for raising this issue!  I can look to resolving these compiler errors.

PlaidShirtPat, 2018-07-27:
You can make this change to supress the warnings if you like:
```diff
diff --git a/examples/platforms/posix/Makefile.am b/examples/platforms/posix/Makefile.am
index 61ffd034..5c7eb2a5 100644
--- a/examples/platforms/posix/Makefile.am
+++ b/examples/platforms/posix/Makefile.am
@@ -35,6 +35,7 @@ libopenthread_posix_a_CPPFLAGS            = \
     -I$(top_srcdir)/examples/platforms      \
     -I$(top_srcdir)/src/core                \
     -D_GNU_SOURCE                           \
+    -Wno-error=class-memaccess              \
     $(NULL)
 
 PLATFORM_SOURCES
```

jwhui, 2018-08-02:
@PlaidShirtPat, thanks for the suggestion.  I went ahead and resolved the warnings rather than disabling them.  I just submitted #2944 to resolve this issue.  Please review.

PR #2944 (2018-08-03): [gcc8] resolve compiler errors
This commit resolves compiler errors generated by gcc8.

This commit also adds gcc8 build checks to travis.

Resolves #2931.
closed 2018-08-03 by jwhui