openthread/openthread #2932: Slaac address lost after restart "thread"

NickNamerr, 2018-07-27:
Tried to restart "thread" on device (it worked before):

> ipaddr
fd11:1111:1122:0:388d:3a46:73b:2ab8
fe80:0:0:0:e8dd:4629:bd81:a4a8
Done
> ifconfig up
Done
> thread start
Done
> ipaddr
2001:db8:8568:b2b5:5618:f8e8:7090:9f67
fd11:1111:1122:0:0:ff:fe00:c808
fd11:1111:1122:0:388d:3a46:73b:2ab8
fe80:0:0:0:e8dd:4629:bd81:a4a8
Done
> thread stop
Done
> ifconfig down
Done
> ipaddr
fd11:1111:1122:0:388d:3a46:73b:2ab8
fe80:0:0:0:e8dd:4629:bd81:a4a8
Done
> ifconfig up
Done
> thread start
Done
> ipaddr
fd11:1111:1122:0:0:ff:fe00:c808
fd11:1111:1122:0:388d:3a46:73b:2ab8
fe80:0:0:0:e8dd:4629:bd81:a4a8
Done
> netdatashow
0314004020010db88568b2b5070211400504fffe3300030f004020010db88569b2b50103fffe00
Done
>

could ping both fd11:1111:1122:0:: addresses from OTBR:
pi@raspberrypi:~/openthread/openthread $ ping6 fd11:1111:1122:0:0:ff:fe00:c808
PING fd11:1111:1122:0:0:ff:fe00:c808(fd11:1111:1122::ff:fe00:c808) 56 data bytes
64 bytes from fd11:1111:1122::ff:fe00:c808: icmp_seq=1 ttl=63 time=94.4 ms
64 bytes from fd11:1111:1122::ff:fe00:c808: icmp_seq=2 ttl=63 time=53.7 ms
64 bytes from fd11:1111:1122::ff:fe00:c808: icmp_seq=3 ttl=63 time=56.1 ms

--- fd11:1111:1122:0:0:ff:fe00:c808 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2001ms
rtt min/avg/max/mdev = 53.707/68.099/94.472/18.674 ms

pi@raspberrypi:~/openthread/openthread $ ping6 fd11:1111:1122:0:388d:3a46:73b:2ab8
PING fd11:1111:1122:0:388d:3a46:73b:2ab8(fd11:1111:1122:0:388d:3a46:73b:2ab8) 56 data bytes
64 bytes from fd11:1111:1122:0:388d:3a46:73b:2ab8: icmp_seq=1 ttl=63 time=115 ms
64 bytes from fd11:1111:1122:0:388d:3a46:73b:2ab8: icmp_seq=2 ttl=63 time=63.5 ms
64 bytes from fd11:1111:1122:0:388d:3a46:73b:2ab8: icmp_seq=3 ttl=63 time=54.1 ms

--- fd11:1111:1122:0:388d:3a46:73b:2ab8 ping statistics ---
4 packets transmitted, 3 received, 25% packet loss, time 3005ms
rtt min/avg/max/mdev = 54.135/77.716/115.422/26.940 ms
pi@raspberrypi:~/openthread/openthread 

1) commit f824343f8cac34863c4267ae96a390da48c86ad3 (noticed on 1075e35327ec97e94404e3ecb0bb2381641423ee)
2) SiLabs EFR32MG13

jwhui, 2018-07-27:
Thanks for raising this issue.  I have reproduced the behavior and will submit a fix soon.

jwhui, 2018-07-30:
I submitted #2938 to resolve this issue.  Please validate/review.

NickNamerr, 2018-08-01:
I checked it, it works:
> ipaddr
2001:db8:8568:b2b5:b5fd:eb36:d2f6:ffb7
fd11:1111:1122:0:0:ff:fe00:6
fe80:0:0:0:94e3:9dff:c548:275c
fd11:1111:1122:0:dc79:4152:e90c:b80b
Done
> thread stop
Done
> ipaddr
2001:db8:8568:b2b5:b5fd:eb36:d2f6:ffb7
fe80:0:0:0:94e3:9dff:c548:275c
fd11:1111:1122:0:dc79:4152:e90c:b80b
Done
> ifconfig down
Done
> ipaddr
fe80:0:0:0:94e3:9dff:c548:275c
fd11:1111:1122:0:dc79:4152:e90c:b80b
Done
> ifconfig up
Done
> ipaddr
fe80:0:0:0:94e3:9dff:c548:275c
fd11:1111:1122:0:dc79:4152:e90c:b80b
Done
> thread start
Done
> ipaddr
2001:db8:8568:b2b5:b5fd:eb36:d2f6:ffb7
fd11:1111:1122:0:0:ff:fe00:6
fe80:0:0:0:94e3:9dff:c548:275c
fd11:1111:1122:0:dc79:4152:e90c:b80b
Done
>

jwhui, 2018-08-01:
@NickNamerr, thanks for validating!

PR #2938 (2018-08-01): [ip6] ensure slaac addresses are added back after reset
This commit makes the following changes:

1. A child that is resynchronizing with its parent after reset always
   requests the network data to ensure that it has the latest version.

2. When autoconfiguring SLAAC addresses in response to a network data update
   always attempt to add the netif address buffer in case the netif
   structure was removed (e.g. due to reset).

Resolves #2932.
closed 2018-08-01 by jwhui