openthread/openthread #2952: UDP value's treatment

blaizi, 2018-08-08:
Hello,

I am currently working on the thread project, on the microprocessor EFR32. I would Like to treat the UDP values that I receive. (for example, set a LED "on" after an UDP message is receive, modify its intensity, send an UDP reply, ...)
However, I'm trying to modify the cli_udp_example.cpp, and I tried many things without success. (like using the header hw_gpio.h, or even use the library available in this github: https://github.com/DannyHavenith/avr_utilities )
My question is: is it currently possible to treat the UDP messages that we receive, and how can we do this ? (I have to precise that I am a beginner in c ++ coding).

there is the kind of error that I have:
``error: 'hw_gpio_toggle' was not declared in this scope
     hw_gpio_toggle(PORTF, 4)``

jwhui, 2018-08-08:
On EFR32, I suggest using the GPIO APIs provided by the SiLabs Gecko SDK.

https://siliconlabs.github.io/Gecko_SDK_Doc/efm32g/html/group__GPIO.html

blaizi, 2018-08-09:
So I just have to create the Header .hpp with the enumeration and functions in the openthread/src/core/ path, and then include this header in the cli_udp_example.cpp file ?

jwhui, 2018-08-09:
Given that the example EFR32 platform library already [compiles in the SiLabs GPIO abstractions](https://github.com/openthread/openthread/blob/101e243372214ab470695506de7c9031f3cb50d1/examples/platforms/efr32/Makefile.am#L105), you should be able to just inluced `em_gpio.h` and call the SiLabs Gecko SDK APIs directly.

blaizi, 2018-08-10:
Ah yes it works, thank you a lot for your help

closed 2018-08-10 by jwhui