openthread/openthread #2959: Support for LAUNCHXL-CC26X2R1

Goran58, 2018-08-14:
**Describe the bug**
Unable to run the CC2652 output can't run on LAUNCHXL-CC26X2R1

**To Reproduce**
build with;  
make -f examples/Makefile-cc2652
Use TI UniFlash tool and download the output file ot-cli-ftd to the launchpad

The download is successfull but no response on the cli connection (minicom to device /dev/ttyACM0)

srickardti, 2018-08-15:
Hey @Goran58, I've got the device running on a debugger and have found that it is spinning in the synchronous wait for the CM0 in `radio.c` on line `968`. This is due to the IEEE overrides to the CM0. I had added the current overrides that were available from SmartRF Studio. That includes the switch to the MCE/RFE patch, which isn't applied.

The easiest way to get up and running is to comment out the switch from the override list, [radio.c line 74](https://github.com/openthread/openthread/blob/master/examples/platforms/cc2652/radio.c#L74). This will result in a less characterized, but functional, radio. I need to spend some time looking into the proper way to apply these patches.

If you would like a characterized version, you can use one of the examples from the [SIMPLELINK-CC26X2-SDK](http://www.ti.com/tool/SIMPLELINK-CC26X2-SDK). That is based on TI-RTOS, so it might not be what you are looking for.

Goran58, 2018-08-15:
Hi @srickardti, thanks for your help. I have today used the examples in the SIMPLELINK SDK and have managed to set up a working thread network that consists of a Border Router and two Router nodes and support for external commissioning with Openthread Commissioning App. Works very well, but it would be nice to be able to use the openthread native examples to do same setup. But I can wait with that until you have applied the patches.

PR #2971 (2018-08-17): [cc13x2_cc26x2] fix runtime issue with RF patching
Resolves #2959

Enable proper application of radio patches. This will fix the halting issue with the radio code on the cc2652.
closed 2018-08-17 by jwhui