openthread/openthread #2967: SAMR21 ASF compile error

TobiasReichert, 2018-08-16:
Hey Guys,

**Describe the bug**
with the newest version of Advanced Software Framework (ASF) you can not compile the SAMR21 Example.

**To Reproduce**
Do the steps on:
[samr21](https://github.com/openthread/openthread/tree/master/examples/platforms/samr21)

**Console/log output**
```
In file included from openthread/examples/../third_party/microchip/include/asf.h:56:0,
                 from openthread/examples/../examples/platforms/samr21/flash.c:34:
openthread/examples/../third_party/microchip/asf/sam0/utils/compiler.h:132:0: error: "__always_inline" redefined [-Werror]
 #  define __always_inline             __attribute__((__always_inline__))
 
In file included from /usr/arm-none-eabi/include/stdio.h:35:0,
                 from openthread/examples/../third_party/microchip/asf/sam0/utils/compiler.h:61,
                 from openthread/examples/../third_party/microchip/include/asf.h:56,
                 from openthread/examples/../examples/platforms/samr21/flash.c:34:
/usr/arm-none-eabi/include/sys/cdefs.h:388:0: note: this is the location of the previous definition
 #define __always_inline __inline__ __attribute__((__always_inline__))
```

**Additional context**
- OS: Linux Mint 18.3 Sylvia
- Linux version: 4.15.0-30-generic 
- asf version: 3.42.0.77
- arm-none-eabi-gcc version: 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

TobiasReichert, 2018-08-16:
asf-3.39 does not work, too.

jwhui, 2018-08-16:
@TobiasReichert, thanks for raising this issue!

This seems to be an issue with the Advanced Software Framework (ASF) itself. In particular, both `asf/sam0/utils/compiler.h` and `arm-none-eabi/include/sys/cdefs.h` (included by the former) define `__always_inline`.

One hack is to apply the following patch to `xdk-asf-3.40.0/sam0/utils/compiler.h`:
```
132c132
< #  define __always_inline             __attribute__((__always_inline__))
---
> //#  define __always_inline             __attribute__((__always_inline__))
```
Basically, avoid defining `__always_inline` since gcc-arm does that for you already.

Can you help report this issue to Microchip to have it properly addressed?

TobiasReichert, 2018-09-07:
Sorry for the late answer.

@jwhui thanks for your answer. I can confirm your fix works. :)

Unfortunately, I have no microchip account to properly forward the bug.

closed 2018-09-07 by TobiasReichert