openthread/openthread #2969: Call UdpOpen and UdpBind fuctions

BJeanjean, 2018-08-16:
Hello,

I am using OpenThread and I would like to call automaticaly the Udp functions at the boot of the Microcontroller (I am using EFR32), then I don't have to use the Cli to send and receive message after creating the port once.
I am currently struggling, I am trying to modifie the Cli_Udp_Example.cpp , without success (I tried for example to call otUdpOpen and otUdpBind functions into UdpExample::UdpExample(Interpreter &aInterpreter), but the Cli doesn't work after flashing the EFR32), 
and I also tried to modify udp_api.cpp with the model of thread_api.cpp (using function otThreadSetAutoStart) but I don't know which argument I have to put in the otUdpOpen and otUdpBind functions.

thank you in advance for your answers.

jwhui, 2018-08-16:
It should be as simple as calling `otUdpOpen()` and `otUdpBind()` as part of the startup code.
- Here is [a _very_ simplistic example](https://gist.github.com/jwhui/5bdce29cd97fe257d3362970f51800e9).
- Take a look at the [UDP API reference](https://openthread.io/reference/group/api-udp) for more information.

BJeanjean, 2018-08-20:
Thank you a lot for your answer.
I have modified the main.c file and added some lines in the handleUdpReceive function and it's now working like I want.

jwhui, 2018-08-22:
@BJeanjean, glad to hear that you got things working!

closed 2018-08-22 by jwhui