openthread/openthread #2980: error: redefinition of 'otError ot::Mle::Mle::GetLeaderAloc(ot::Ip6::Address&) const

NickNamerr, 2018-08-21:
commit af2bfc1cc4a9b48e2c81b63645b45d67ac1ad11b

/home/user/openthread/openthread/examples/../src/core/thread/mle.cpp:993:9: error: redefinition of 'otError ot::Mle::Mle::GetLeaderAloc(ot::Ip6::Address&) const'
 otError Mle::GetLeaderAloc(Ip6::Address &aAddress) const
         ^~~
In file included from /home/user/openthread/openthread/examples/../src/core/thread/mle.cpp:36:0:
/home/user/openthread/openthread/examples/../src/core/thread/mle.hpp:857:13: note: 'otError ot::Mle::Mle::GetLeaderAloc(ot::Ip6::Address&) const' previously defined here
     otError GetLeaderAloc(Ip6::Address &aAddress) const { return GetAlocAddress(aAddress, kAloc16Leader); }
             ^~~~~~~~~~~~~
  CXX      thread/libopenthread_ftd_a-network_diagnostic.o

bukepo, 2018-08-21:
This commit didn't change mle.hpp or mle.cpp. What version are you compiling?

librasungirl, 2018-08-21:
For a clean af2bfc1, the Line 993 in mle.cpp should be `otError Mle::GetAlocAddress(Ip6::Address &aAddress, uint16_t aAloc16) const`, other than `GetLeaderAloc()`.

jwhui, 2018-08-22:
@NickNamerr, as @librasungirl noted, L993 of mle.cpp does not match with the error you included above. Can you ensure that you have a consistent and clean version of the OpenThread source files?

https://github.com/openthread/openthread/blob/af2bfc1cc4a9b48e2c81b63645b45d67ac1ad11b/src/core/thread/mle.cpp#L993

NickNamerr, 2018-08-23:
Please check "mle.hpp":
https://github.com/openthread/openthread/blob/af2bfc1cc4a9b48e2c81b63645b45d67ac1ad11b/src/core/thread/mle.hpp#L857

jwhui, 2018-08-23:
I don't yet see the problem. The implementation of `GetLeaderAloc()` makes a call to `GetAlocAddress()`. Can you clarify how the two lines of code below are in conflict with each other?

https://github.com/openthread/openthread/blob/af2bfc1cc4a9b48e2c81b63645b45d67ac1ad11b/src/core/thread/mle.cpp#L993

https://github.com/openthread/openthread/blob/af2bfc1cc4a9b48e2c81b63645b45d67ac1ad11b/src/core/thread/mle.hpp#L857

closed 2018-08-27 by jwhui