openthread/openthread #2981: Debugging in target environnement

tototrix, 2018-08-21:
Hi everyone,
I tried OpenThread with my Silabs EFR32 and it's working great. To go further i would like to code and debug my project in Simplicity Studio with breakpoints and so on.
Is it possible to do that ? 
I fear that the only debug capabilities of OpenThread are with debug log. 

Thanks in advance for your feedback.

jwhui, 2018-08-22:
I am not familiar with Silicon Labs' Simplicity Studio IDE. However, I assume all you need to do is include the OpenThread source files in a Simplicity Studio project. Using the build+debug command should just work.

Alternatively, you could just use the ARM gcc toolchain (i.e. `arm-none-eabi-gdb`) to directly debug without an IDE.

tototrix, 2018-08-22:
Thank you for your quick reply.

Unfortunately i am not familiar with command line debugging:

- Have you some documentation/tutorials to read ?
- Is-it possible to do step-by-step debugging ?
- Possible to put breakpoints ?
- To view variables values ( watch values in IDE) ?

Thank you again for your help.

hjian2017, 2018-08-22:
I am not familiar with Silicon Labs' Simplicity Studio IDE, but I think you can debug in Simplicity Studio with breakpoints .

jwhui, 2018-08-28:
You can find some resources by searching around the web. One such resource: https://mcuoneclipse.com/2015/03/25/command-line-programming-and-debugging-with-gdb/

closed 2018-08-28 by jwhui

kentmartin99, 2020-10-13:
I have tried replacing the thread files (under gecko_sdk_3.0.0/util/thirdparty/openthread/ with the files from the git openthread repo and it will not build. I manually looked at the folders in SS% and found the folders in openthread and made another set of folder with the same structure as in SS5 and substituted that folder into the project. It does not build with many errors.
How can we update the openthread files in SS5 to those from openthread github?

jwhui, 2020-10-14:
> I have tried replacing the thread files (under gecko_sdk_3.0.0/util/thirdparty/openthread/ with the files from the git openthread repo and it will not build. I manually looked at the folders in SS% [...]

@kentmartin99 , I would reach out to Silicon Labs directly with this question.

kentmartin99, 2020-10-14:
Jonathan,
Thanks, I have, their first response was to build it under linux.
Unfortunately the linux build environment lacks many things and is
abhorrent to use from windows and not easy to adjust for new boards or new
applications. Unfortunately my machine has not taken well to linux vm's ( i
think it's an AMD issue) so I am stuck with WSL2 for now.
I will get some stuff going under SS5 and possibly try to port those
changes back to openthread repo later. It's a long way round but the
environment is just so awful I don't think it will get done any other way.
It would be nice if some effort was put into making this easier to use.
Cheers.