openthread/openthread #2985: Problem about Dataset Delay Timer Management

swordinrock123, 2018-08-22:
Hello,

When I was trying the harness certificate test in NCP mode:
9.2.18 RollingBackTheAciveTimestampWithPendingCommissioningDataset(Router and ED).
I got failure results and then I did some analysis, I guess I found a problem about the Dataset Delay Timer Management, 

According to the specification(8.4.3.4 Delay Timer Management):
When a Thread Device's Delay Timer reaches zero, the Thread Device compares Active Operational Dataset Active Timestamp with the Pending Operational Dataset Active Timestamp. If the Pending Operational Dataset Active Timestamp is newer **or the Pending Operational Dataset contains a new Network Master Key**, the Thread Device MUST:
1. Replace the Active Operational Dataset with the Pending Operational Dataset, removing the Delay Timer and Pending Timestamp at the same time.
2. Commit the new Active Operational Dataset to non-volatile memory.
3. Completely remove the Pending Operational Dataset.

The case 9.2.18 uses a new Network Master Key(00:11:22:33:44:55:66:77:88:99:aa:bb:cc:dd:ee:ee) in pending dataset, so it is supposed that the pending operational dataset will take effect after delay timer timeout.

However, I checked the API otError DatasetManager::Set(const Dataset &aDataset)
https://github.com/openthread/openthread/blob/master/src/core/meshcop/dataset_manager.cpp#L147
I found that it only cares about the timestamp, but doesn't take masterkey into consideration.

The result is, before the delay timer timeout, pending dataset is:
Thread:PendingDataset = [
        "Dataset:ActiveTimestamp          =  0x0000000000000014"
        "Dataset:PendingTimestamp         =  0x0000000000000014"
        "Dataset:Channel                  =  15"
        "Dataset:NetworkName              =  "MyHouse""
        "Dataset:PanId                    =  0xFACE"
        "Dataset:ExtendedPanId            =  0x000DB8000000"
        "Dataset:MasterKey                =  [00112233445566778899AABBCCDDEE**EE**]"
        "Dataset:MeshLocalPrefix          =  fd00:db8::/64"
        "Dataset:Delay                    =  19198"
        "Dataset:ChannelMaskPage0         =  0x07FFF800"
        "Dataset:PSKc                     =  [10B95765596AB9D0B86CEBDD0FA24DA3]"
        "Dataset:SecPolicy:KeyRotation    =  3600"
        "Dataset:SecPolicy:Flags          =  0xF8"
]

After delay timer timeout,  the pending dataset is removed but active dataset is still unchanged.
wpanctl:wpan0> get Thread:PendingDataset
Thread:PendingDataset = []
wpanctl:wpan0> get Thread:ActiveDataset
Thread:ActiveDataset = [
        "Dataset:ActiveTimestamp          =  0x0000000000004E20"
        "Dataset:Channel                  =  15"
        "Dataset:NetworkName              =  "GRL""
        "Dataset:PanId                    =  0xFACE"
        "Dataset:ExtendedPanId            =  0x000DB8000000"
        "Dataset:MasterKey                =  [00112233445566778899AABBCCDDEE**FF**]"
        "Dataset:MeshLocalPrefix          =  fd00:db8::/64"
        "Dataset:ChannelMaskPage0         =  0x07FFF800"
        "Dataset:PSKc                     =  [10B95765596AB9D0B86CEBDD0FA24DA3]"
        "Dataset:SecPolicy:KeyRotation    =  3600"
        "Dataset:SecPolicy:Flags          =  0xF8"
]

In my opinion, it is against the specification, active dataset should be replaced by pending dataset, hope to know your thoughts about this.
Thanks!

librasungirl, 2018-08-22:
@swordinrock123 , thanks a lot for raising this issue and excellent analysis. Agree that there is indeed a bug as you point out.
 
The phenomenon caused by this issue would be that OT device would have the dataset (with activetimestamp 20) in use, but have dataset (activetimestamp 0x4e20) in local, thus it would keep sending MGMT_SET_ACTIVE.req to Leader,  however get response with Reject status.  
In other words, after delay timer expires, the devices would indeed switch to new masterkey, and traffic would be encrypted with the new key.  The weird thing is that existing test script in Test Harness (at least in V49.1) looks to me only check the ping traffic after switching to new master key which should not be impacted by this bug.

BTW, may I know which Test Harness version you are using?

librasungirl, 2018-08-22:
@swordinrock123 , I have proposed a fix as in #2988 , please help to have a review. Thanks.

swordinrock123, 2018-08-23:
@librasungirl I just wrote a comment about your PR.
I'm using the R1.3.0 Harness tool, which has v48.0 internal, I'm not sure about the root cause of 9.2.18, but if I modify the DatasetManager::Set and have mLocal.Set(aDataset) always work, I can pass the case.
Anyway, I will check more about it, and when your solution to this bug is ready, I will test it. Thanks!

swordinrock123, 2018-08-23:
@librasungirl

swordinrock123, 2018-08-23:
@librasungirl ， Hi， thanks for your quick response, I just tested your solution, it can help me pass the 9.2.18(Router and ED), and it looks good to me about your solution. I may need more research to know why this case is impacted by this bug.
PS: My harness tool version is R 1.3.0(v48.0)

librasungirl, 2018-08-24:
@swordinrock123 I have pushed another proposal in #2988 to fix the issue. Would you please help to test it? Thanks.

jwhui, 2018-08-28:
Resolved by #2988.

closed 2018-08-28 by jwhui

swordinrock123, 2018-08-29:
@librasungirl Sorry about replying late, I just use your new proposal and tested the 9.2.18, it works fine, Thank you!

Ahmedkalnaggar, 2018-09-08:
@librasungirl @jwhui I am running a CLI app not NCP as the main issue.

librasungirl, 2018-09-08:
@Ahmedkalnaggar , when apply pending dataset, generally `mRadioChannel` would be updated immediately in Mac layer.
https://github.com/openthread/openthread/blob/master/src/core/mac/mac.cpp#L535
That is, both transmission and receive should be on the new channel.

Would you please provide more information? e.g your certification log, or your cert environment (platform, OT commit, Test Harness Version)?

LuDuda, 2018-09-08:
@librasungirl  I run Thread Harness v48.2 yesterday (it's still running) on current OT master and i also see failures in SED_9_2_6, ED_9_2_6 and Router_9_2_6. I will take a closer look on this at Monday, but in all cases it seems DUT doesnt switch to the second channel.

The one change I did was replacing NXP with OpenThread for SED role in TopologyConfig.txt due to TESTPLAN-380 issue (but this is irrelevant).

[Harness PDFs and Wireshark logs for 9_2_6](https://github.com/openthread/openthread/files/2363585/Test_9_2_6.zip)

Besides, also Router_9_2_10 failed. Do you know, if some tests are supposed to fail in v48.2 and current OT master?

LuDuda, 2018-09-08:
I think i found the problem of 9.2.6.

It was introduced in #2838 more precisely in this [line](https://github.com/abtink/openthread/blob/master/src/core/mac/mac.cpp#L526). This prevents device to change channel defined in Pending Dataset.

Logs from device says:
` 0> [0000226249] -MESH-CP-: DatasetManager::ApplyConfiguration() Failed to set channel to 21 (InvalidArgs)`

The problem is that channel mask is set to 0x00008000 (only 16th channel is enabled), then commissioner only updates channel to 21 without touching channel mask.

Thread Specification says: 

> Channel Mask: the set of PHY-layer channels that a device should use when performing network discovery using Discovery Messages or MLE Announce frames as well as when performing MAC ED or active scans.

I don't see however any text preventing to changle channel without modyfing channel mask, so probably we should remove those checks from `SetPanChannel` and `SetRadioChannel` methods.

Thoughts?

librasungirl, 2018-09-09:
Thanks @Ahmedkalnaggar for raising this issue and @LuDuda for findings.

We didn't find this issue in our internal regression on recent commit as on OT-only testbed, the default channel mask is '001fffe0' (11~26), the same behavior as ARM or SiLabs, and different from NXP which has channel mask, including only the channel it works on. #2838 is indeed the root cause for the failings you met.

Agree that there is no clear text in spec, however personally I think there is some sense in #2838. 
Since it is agreed that Discovery or Announce could only search available networks on the channels allowed by channel mask, it means that the device should only work on one of the allowed channels.

Thoughts? @jwhui  @LuDuda @Ahmedkalnaggar

Ahmedkalnaggar, 2018-09-09:
Thanks @librasungirl  for the response and @LuDuda for the analysis 👍 
I could confirm now that the problem with channel mask from Leader which is NXP in normal Topology Config file:
The channel mask is including only the old channel as @librasungirl  mentioned before. It's not updated with a proper mask for the new channel :
[image]
but if we changed the the Leader to ARM, this would fix the issue without editing the code. As the channel mask in ARM always transmitted with the default value :
[image]

Ahmedkalnaggar, 2018-09-09:
I have opened issue [TSTBED-106](https://threadgroup.atlassian.net/browse/TSTBED-106) on Thread group. I referenced our conversion here.

abtink, 2018-09-10:
From analysis above by @LuDuda  and @librasungirl , it looks like a pending dataset is being applied which tries to change the channel to one that is not included in the channel mask (the channel mask the same dataset  or earlier one?). If that's the case, then the fact that we reject the new channel, in my opinion, is correct behavior. 

If a dataset is trying to change the channel, it should also change the channel mask at the same time.

That said, I do indeed see an issue with the current implementation of `dataset.cpp`, where if the new dataset tries to change both channel and channel mask, the order of TLVs and how they are processed  matter (here [dataset.cpp](https://github.com/openthread/openthread/blob/master/src/core/meshcop/dataset.cpp#L536)). We should ensure that if there is channel mask entry in the dataset, it's applied before we set the channel. I have a PR to address this: https://github.com/openthread/openthread/pull/3057