openthread/openthread #3056: [MAC] Channel mask check causing failure in Case 9.2.6

Ahmedkalnaggar, 2018-09-10:
**Describe the bug**
There is a problem in Latest Openthread versions with Test Case 9.2.6 (Router, ED and SED) it always fails in last step : 
> Step 26: Verify connectivity by sending an ICMPv6 Echo Request to the DUT mesh local address on the (new) Secondary channel, The DUT MUST respond with an ICMPv6 Echo Reply. 

After some analysis and discussions we found that NXP (which is the Leader in default Topology Config file) send a channel mask with only old channel, not updated with a mask for the new (Secondary) channel. This leads the router (which is the DUT in test case) to refuse moving to the secondary channel as it's not accepted by channel mask due to the edits made in this issue #2838 in this [line](https://github.com/openthread/openthread/blob/master/src/core/mac/mac.cpp#L526) and this [line](https://github.com/openthread/openthread/blob/master/src/core/mac/mac.cpp#L550) . 
After opening Issue in Thread Group [TSTBED-106](https://threadgroup.atlassian.net/browse/TSTBED-106) stating the problem, they replied that this check is too specific : 
`As per section 8.7.5.8 , all other TLVs that are not included in Pending Set are copied from the Active Operational Dataset, so no behavior where Leader modified any parameters based on what it receives is specified.
Also, chapter 8.4.3.4 defines how Pending Operational Dataset should be installed after Delay Timer expires, the Active operational Dataset must be replaced by the Pending Operational Dataset and the network should start function with the new Operational Dataset, nothing related to the use of Scan Mask is specified here. `

**Note**
Replacing the NXP leader with any other vendor, like ARM for example will overcome the issue.  As the channel mask in ARM always transmitted with the default value "0x001fffe0".

Ahmedkalnaggar, 2018-09-10:
Conversation started here #2985 .

jwhui, 2018-09-10:
From the conversation in #2985, I believe we need to revert the Channel Mask check. As @LuDuda mentions, the Thread 1.1.1 Specification makes no mention of restricting channel changes based on the Channel Mask. If the intended behavior is for the Channel Mask to also restrict channel changes, we would first need to make an update to the Thread specification and then the Thread Test specification.

/cc @librasungirl @LuDuda @abtink

abtink, 2018-09-10:
In my opinion, the issue is with definition of "Channel Mask" TLV in dataset and that it is not intended to be used as "supported channels".

The solution I would suggest is as follows: 
- We don't tie the MAC  "supported channel mask" with the "channel mask" TLV in dataset. 
- Basically treat them as two (unrelated for now) network parameters.
- And we keep the "supported channel" at MAC layer and the check at MAC layer for it to be valid.

I would rather we keep the "supported channel" and the check in MAC layer as it serves an important purpose. It ensues that a device would not move/send on a restricted channel due to regulatory requirements of the region device is operating in.

jwhui, 2018-09-10:
@abtink, your solution makes sense. As you noted, the Thread 1.1.1 Specification does not currently specify the Channel Mask TLV as a restriction on the channel that may be used for normal data communication.

abtink, 2018-09-10:
Submitted PR https://github.com/openthread/openthread/pull/3058 with solution above.

jwhui, 2018-09-12:
Resolved by #3058.

closed 2018-09-12 by jwhui