openthread/openthread #3079: Micropython openthread support

CraigZych, 2018-09-17:
I am trying to make a working example of openthread running on micropython on the nrf52840. 
I opened an issue to discuss this on the micropython github page but figured there may be interest here as well. 

With the ease of development on micropython porting this over may bring a considerable audience to openthread.

Pycom has already ported openthread to micropython but specifically on the lora radio, this provides incrediable range but brings latency issues.

If anyone is interested in helping to bring openthread to the micropython platform please let me know.

jwhui, 2018-09-17:
@CraigZych, sounds like a great initiative!

jwhui, 2018-10-11:
@CraigZych, are you actively working on MicroPython support? Do you have any updates to provide?

CraigZych, 2018-10-14:
I am actively working on it but with limited time. 

I'm still very interested if anyone is interested in helping. 

I am currently working the angle that the best way is through the micropython zephyr port since it has openthread support. I had tried originally with the nrf port but this will open up many more board options.

CraigZych, 2018-11-02:
@jwhui @rlubos 

Any chance either of you can provide some insight into a stumbling block I'm having.

I'm trying to use the zephyr port of micropython https://github.com/pfalcon/micropython

Clone that repo and cd into ports/zephyr

If I copy the overlay-ot.conf (from this project zephyr/samples/net/echo-client) file and add the -DOVERLAY_CONFIG parameter from that example to line 105 of the MAKEFILE it attempts to build in openthread but doesn't actually download or include it.

It throws the same errors I get if I add the DOVERLAY_CONFIG parameter without deleting the build directory in the zephyr/samples/net/echo-client directory of this project. I'm guessing I have some cache or other issue.

I'm hoping to get this going and this is the first major step before I can start implementing the necessary components to expose the openthread functionality into micropython.

I think if I can get this going (with some help) it will significantly broaden the reach of openthread into the maker community by creating a dead simple usage environment. Hopefully someone can help.

bmeisels, 2018-11-02:
@CraigZych 
I have been playing with this today and i managed to remove the errors.
You are missing some configuration that the zephyr openthread  module is dependent on such as CONFIG_FLASH.
See my last message on https://github.com/micropython/micropython/issues/4098

rlubos, 2018-11-05:
@CraigZych 
Just to add my 2 cents here - due to config file refactorization applied some time age to the echo samples in Zephyr, they now use 3 config files to build OpenThread for nRF52840 - `prj.conf`, `boards/nrf52840_pca10056.conf` and `overlay-ot.conf`. As `nrf52840_pca10056.conf` is used by both, OpenThread and raw 802.15.4 configuration of this sample, some configs vital for OpenThread might be present there as well (as already pointed out by @bmeisels).

rlubos, 2018-11-06:
@Craigzyc 
Unfortuantely nobody was working (yet) on improving the Zephyr port for OpenThread, so the static configuration of OpenThread parameters is the only possible option available out-of-the box.

CraigZych, 2018-11-06:
@rlubos

I know i had tried to change the panid through the cli and it didnt work. Is that due to it statically coded?

rlubos, 2018-11-07:
Well partially because of this. You can't modify PANID while network is up and running, and as it starts in Zephyr automatically, that's what's blocking you. If you have issued `thread stop` command first, you should be able to modify network parameters. Note, that you have to re-enable the network with `thread start` afterwards.

Craigzyc, 2018-11-07:
Ok. I totally should have remembered that from playing with openthread before.

So even though we can't stop the auto initialization we can `thread stop` and setup the network the way we need it.

@rlubos 
Can the cli be accessed from within code in the current implementation or is that only in the console manually?

rlubos, 2018-11-07:
From what I know the OpenThread APIs are not exposed to the Zephyr application for now. It should be doable, still it's not something available out-of-the box.

Craigzyc, 2018-11-07:
Ok, Thank you.

From what I saw in the comments on the zephyr issues it sounded like you were working on the openthread implementation but thats obviously not the case. Thanks for your help

jwhui, 2019-01-17:
@Craigzyc, any updates?

jwhui, 2019-02-01:
Closing stale issue.

closed 2019-02-01 by jwhui