openthread/openthread #3087: nrf52840-dongle is unable to work as NCP

IotPanorama, 2018-09-18:
**Bug details**
nrf52840 is not being able to work as NCP device. It throws initialization fault with border router. (we have tried with both RPI and linux machine)

**To Reproduce**
Compile and burn firmware of ncp-ftd (along with bootloader) in to nrf52840 dongle by using nRF connect Desktop app. with that set up try to use the dongle as NCP device with boarder router.

**Expected behavior**
It should have successfully initialized.

**Console/log output**
[image]

jwhui, 2018-09-18:
Can you provide more information to reproduce?

For example:
- Commit ID for NCP firmware.
- Commit ID for wpantund.
- Build command/configuration you used for NCP firmware.

IotPanorama, 2018-09-19:
wpantund, commit id: 6752c40c50f5c556c3b0d150c250a0d1a0a82d32
NCP firmware commit id: 3da8087492138fbeca579318efab8a9506d91f0e
command used to build NCP Firmware. 
1. make -f examples/Makefile-nrf52840 BORDER_AGENT=1 BORDER_ROUTER=1 COMMISSIONER=1 UDP_PROXY=1 USB=1
2. make -f examples/Makefile-nrf52840 COMMISSIONER=1 USB=1

jwhui, 2018-09-20:
The wpantund commit ID is over a year old. I could not find the NCP commit id within the openthread/openthread repo.

Can you try again with the latest master branches of both openthread/wpantund and openthread/openthread?

IotPanorama, 2018-09-20:
Ncp commit id, I have mentioned is actually commit id of openthread itself. If you guys have different commits for ncp firmware in particular then let me know. I will test that latest repository of both openthread and wpantund and will let you know about updates.

jwhui, 2018-09-20:
As mentioned, commit id (https://github.com/openthread/openthread/commit/3da8087) does not exist - following the GitHub link results in an error.

Hope the latest master head of both openthread and wpantund will resolve your issue.

e-rk, 2018-09-20:
Hi IotPanorama,
When building the examples for the dongle with bootloader use the BOOTLOADER=1 flag along with USB=1 when running make. Your command should look like this:

    make -f examples/Makefile-nrf52840 COMMISSIONER=1 USB=1 BOOTLOADER=1

Before that however you have to clean all previously generated files using this command:

    make -f examples/Makefile-nrf52840 clean

You may also have to flash the bootloader as it might have got overwritten when flashing the hex without the bootloader flag. In nRF Connect verify that your hex begins at the 0x00001000 address.

If that doesn't work try making a fully featured build. More info can be found [here](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.thread_zigbee.v1.0.0%2Fthread_intro.html&cp=4_2_0_1_0_4&anchor=thread_qsg_libraries). Just remember to replace TMF_PROXY with UDP_PROXY.

Please let me know if it helped.

jwhui, 2018-09-27:
@IotPanorama, any update on this issue?

e-rk, 2018-10-26:
Hi,
Can you please try checking out to the current master branch? Recently a bug with USB communication was fixed. Also try building the examples with these build flags:

        make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1 USB=1 BOOTLOADER=1
Hope it helped.

riedlse, 2018-11-16:
I just got a few nRF dongles, I can't seem to get it programmed using the nRF Connect programmer. I tried the normal border router build and the one above. I select the file, but I can't write or erase. nRF connect 2.6.1 programmer 1.0.0

pszkotak, 2018-11-19:
@arn-soft , on the last screenshot you provided it seems that in the middle of October you were using the month-old openthread commit. It is not a good idea for two reasons.
1. There is active development going on, including important bug fixes.
2. In a case when wpantund or openthread is much older than the other it may turn out that some change in the spinel protocol used to communicate between them that took place. It used to cause incompatibility issues in the past. Please follow advice given by @jwhui and use latest wpantund and openthread.

In case you find border router setup from scratch too complicated you may want to check the 
OpenThread Border Router build documentation https://openthread.io/guides/border-router/raspberry-pi-3b or even use the precompiled and preconfigured RPi 3B image provided by Nordic Semiconductor https://www.nordicsemi.com/eng/Products/nRF5-SDK-for-Thread - besides the image there is also an extensive documentation available.

@riedlse, please make sure that the dongle entered the bootloader (indicated by pulsating red LED).

Then build the binary with the comamnd mentioned by @e-rk .
Next you need to execute the:
`arm-none-eabi-objcopy output/nrf52840/bin/ot-cli-ftd -O ihex ot-cli-ftd.hex`
And use the ot-cli-ftd.hex with the programmer app.

Kind regards,
Piotr

riedlse, 2018-11-20:
red led is flashing, bring up nRF tools on the mac, open the programmer app, finds the device in DFU mode, load the hex file... Still can't program, do I have to add the DFU red code area or something to the download?
[image]

pszkotak, 2018-11-20:
It seems that the application overlaps the bootloader - small and big orange sections.
Have you built the firmware with the BOOTLOADER=1 flag?

Please run the following to be sure
1. `cd openthread`
2. `make -f examples/Makefile-nrf52840 clean`
3. `make -f examples/Makefile-nrf52840 USB=1 BOOTLOADER=1`
4. `arm-none-eabi-objcopy output/nrf52840/bin/ot-cli-ftd -O ihex cli.hex`
5. Load the `cli.hex` in the programmer app.

riedlse, 2018-11-20:
make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1 USB=1 BOOTLOADER=1

Used the make file from above, I will try the shorter version...

pszkotak, 2018-11-20:
The amount of used parameters should not matter. The `make ... clean` does.

riedlse, 2018-11-20:
Nope, although instead of orange it shows up as green this time... Ran the make clean, used to doing that for open thread whenever I build.

It does look like it starts above 0, but not high enough

[image]

riedlse, 2018-11-20:
sorry wrong screen shot
[image]

riedlse, 2018-11-20:
Is there a command line way to use nrfjprog with this dfu dongle? it obviously does not have the debugger loaded.

pszkotak, 2018-11-20:
Actually, when you would hover over the orange area you would see that it ends at 0x1000 as well.
The image can be written. Looks good to me.

pszkotak, 2018-11-20:
There is no way to use the nrfjprog to flash the dongle without the J-Link.
The only command line way I am know is use of the USB DFU with the nrfutil.
It is more troublesome than nrfconnect as it requires generation of the app dfu package.
http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.tools%2Fdita%2Ftools%2Fnrfutil%2Fnrfutil_dfu_serial_usb.html

closed 2019-01-03 by jwhui

emenezes, 2019-03-29:
Hi @jwhui, I know that this issue is closed and about 4 months old but I have the same problem here. Since it ended up straying a bit from the original problem and @riedlse did not follow up on the resolution, I'm asking again.

> Can you provide more information to reproduce?
> 
> For example:
> 
> * Commit ID for NCP firmware.
> * Commit ID for wpantund.
> * Build command/configuration you used for NCP firmware.

```
menezes@prime:~/wpantund$ git show
commit 42f9b6e553ebb0ed652ea8212b53c652865da19a (HEAD -> master, origin/master, origin/HEAD)
Author: Yakun Xu <xyk@google.com>
Date:   Sat Mar 30 02:08:09 2019 +0800

menezes@prime:~/openthread$ git show
commit 19544490e37a88642f5e016388038a9ac9def174 (HEAD -> master, origin/master, origin/HEAD)
Author: Jonathan Hui <jonhui@google.com>
Date:   Thu Mar 28 22:29:58 2019 -0700

menezes@prime:~/openthread$ make -f examples/Makefile-nrf52840 COMMISSIONER=1 JOINER=1 COAP=1 DNS_CLIENT=1 MTD_NETDIAG=1 BORDER_ROUTER=1 MAC_FILTER=1 UDP_PROXY=1 BORDER_AGENT=1 LINK_RAW=1 SERVICE=1 USB=1 BOOTLOADER=1
menezes@prime:~/openthread$ arm-none-eabi-objcopy output/nrf52840/bin/ot-ncp-ftd -O ihex output/nrf52840/bin/ot-ncp-ftd.hex
```

I flashed the hex file using nrfconnect. 

The really funny thing is that I can connect to the NCP using pyspinel and the command _state_ returns _leader_ as it should. I also got working the NCP using the posix socket (ot-cli OR wpantund + ot-ncp) with the ot-ncp-radio firmware flashed in the dongle.

dmesg also seems right when I remove and insert back the dongle:

```
[mar29 19:37] usb 3-3: USB disconnect, device number 19
[  +7,976417] usb 3-3: new full-speed USB device number 20 using xhci_hcd
[  +0,150903] usb 3-3: New USB device found, idVendor=1915, idProduct=cafe
[  +0,000008] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000004] usb 3-3: Product: nRF52840 OpenThread Device
[  +0,000005] usb 3-3: Manufacturer: Nordic Semiconductor
[  +0,000004] usb 3-3: SerialNumber: E20EFXXXXXXX
[  +0,002014] cdc_acm 3-3:1.1: ttyACM0: USB ACM device
```

My wpantund.conf:

```
Config:NCP:SocketPath "/dev/ttyACM0"
Config:NCP:DriverName spinel
Daemon:SyslogMask "all"
```

Output from wpantund after reseting the NCP using wpanctl:

```
menezes@prime:~/wpantund$ sudo wpantund
wpantund[28434]: Starting wpantund 0.08.00d (Mar 29 2019 15:25:58) . . .
wpantund[28434]:     SOURCE_VERSION = 0.07.01-286-g42f9b6e
wpantund[28434]:     BUILD_VERSION = 0.07.01-286-g42f9b6e
wpantund[28434]: Configuration file "/etc/wpantund.conf" read.
wpantund[28434]: set-config-param: "Daemon:SyslogMask" set succeded
wpantund[28434]: Stat (autolog): Tx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[28434]: Stat (autolog): Tx history is empty
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog): Rx: 0 packets (0 udp, 0 tcp, 0 icmp6) -- 0 bytes
wpantund[28434]: Stat (autolog): Rx history is empty
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog): NCP state history is empty.
wpantund[28434]: Stat (autolog):
wpantund[28434]: Stat (autolog):
wpantund[28434]: Opening tun interface socket with name "wpan0"
wpantund[28434]: Getting DBus connection
[... 149 lines cut ...]
wpantund[28434]: [NCP->] (0) PROP_VALUE_IS(IPV6_MULTICAST_ADDRESS_TABLE) [1000FF330040FDDE...]
wpantund[28434]: [-NCP-]: Multicast IPv6 address [0] "ff33:40:fdde:ad00:beef::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [1] "ff32:40:fdde:ad00:beef::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [2] "ff02::2"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [3] "ff03::2"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [4] "ff02::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [5] "ff03::1"
wpantund[28434]: [-NCP-]: Multicast IPv6 address [6] "ff03::fc"
wpantund[28434]: Resetting and trying again... (retry 3)
wpantund[28434]: [->NCP] (6) RESET
wpantund[28434]: [NCP->] (0) PROP_VALUE_IS(LAST_STATUS) [70]
wpantund[28434]: [-NCP-]: Last status (RESET_POWER_ON, 112)
wpantund[28434]: [-NCP-]: NCP was reset (RESET_POWER_ON, 112)
wpantund[28434]: [->NCP] (7) PROP_VALUE_GET(PROTOCOL_VERSION)
wpantund[28434]: [NCP->] (7) PROP_VALUE_IS(PROTOCOL_VERSION) [0403]
wpantund[28434]: [->NCP] (8) PROP_VALUE_GET(NET_STACK_UP)
wpantund[28434]: [NCP->] (8) PROP_VALUE_IS(NET_STACK_UP) [01]
wpantund[28434]: [-NCP-]: Stack is up
wpantund[28434]: State change: "uninitialized" -> "associating"
wpantund[28434]: SpinelNCPInstance-Protothreads.cpp:430: Requirement Failed (!ncp_state_is_joining(get_ncp_state()))
wpantund[28434]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[28434]: State change: "associating" -> "uninitialized:fault"
wpantund[28434]: DBus Sending Interface Added Signal for wpan0
^C
```