openthread/openthread #3094: "neighbor table" or "neighbor list" shows only routers.

NickNamerr, 2018-09-24:
regarding https://github.com/openthread/openthread/tree/master/src/cli#neighbor-list
commands from subject must show REED as well.

> neighbor list
0x0000 0x1000 0x1400 0x3000 0x3800 0x3c00 0x4400 0x5800 0x6c00 0x7800 0xa800 0xb000 0xb400 0xb800 0xcc00 0xec00
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|S|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+-+------------------+
|   R  | 0x0000 |  26 |      -63 |       -63 |1|0|1|1| 9299ab86051a2a8a |
|   R  | 0x1000 |  19 |      -74 |       -74 |1|0|1|1| f6821f000821be33 |
|   R  | 0x1400 |  25 |      -60 |       -60 |1|0|1|1| be281972f19fe684 |
|   R  | 0x3000 |  20 |      -52 |       -52 |1|0|1|1| 461baba7c4c5343f |
|   R  | 0x3800 |  27 |      -77 |       -77 |1|0|1|1| c27c35c22540faff |
|   R  | 0x3c00 |  23 |      -64 |       -64 |1|0|1|1| ca68fe2981e9647b |
|   R  | 0x4400 |  24 |      -70 |       -70 |1|0|1|1| daeabd7f10574e09 |
|   R  | 0x5800 |  24 |      -77 |       -77 |1|0|1|1| 9a59e8fe4b807196 |
|   R  | 0x6c00 |  30 |      -84 |       -84 |1|0|1|1| fe0583c48c6f28c1 |
|   R  | 0x7800 |  24 |      -76 |       -76 |1|0|1|1| ead4faaa03d8e4ce |
|   R  | 0x8400 |  17 |      -81 |       -81 |1|0|1|1| 7a13a4b31e2d3dd0 |
|   R  | 0xa800 |  26 |      -70 |       -70 |1
>

librasungirl, 2018-09-24:
@NickNamerr ,  In existing implementation, `MleRouter::GetNextNeighborInfo()` first retrieves valid child info and then neighboring router info as 
https://github.com/openthread/openthread/blob/master/src/core/thread/mle_router.cpp#L3610

So if REED is the device's neighbor (child), it should be shown by the `neighbor list/table` cmd

Below is the sample output on Leader under the topology Router-Leader-REED, fyi

```
> neighbor list
0x5402 0xc400
Done
> neighbor table
| Role | RLOC16 | Age | Avg RSSI | Last RSSI |R|S|D|N| Extended MAC     |
+------+--------+-----+----------+-----------+-+-+-+-+------------------+
|   C  | 0x5402 |  13 |      -20 |       -20 |1|1|1|1| b2c3b9e499401f1b |
|   R  | 0xc400 |   3 |      -20 |       -20 |1|0|1|1| 4a10da869bae6805 |
```

NickNamerr, 2018-09-24:
Thanks for explanation.
So, this is list of neighbor routers and childs of device itself, is it?

librasungirl, 2018-09-24:
Yes. You are correct.
In fact, for REED, it only maintains the one way frame-count synchronization with neighboring non-parent routers, to process incoming advertisement for efficiency (Thread 1.1 spec Section 4.7.7.4), however only its parent would maintain the bi-direction link with the REED, and the other neighboring routers would not take it as valid neighbor.

closed 2018-09-24 by NickNamerr