openthread/openthread #3095: execute "wpanctl get" - causes NCP stuck

NickNamerr, 2018-09-24:
After executes 4-5 times in the row command "wpanctl get" it stuck on showing parameter "Thread:NeighborTable". Here is some logs for better understanding the issue:

"wpanctl status" :
wpan0 => [
        "NCP:State" => "associated"
        "Daemon:Enabled" => true
        "NCP:Version" => "OPENTHREAD/20170716-00904-g06bff963-dirty; EFR32; Sep 24 2018 15:34:21"
        "Daemon:Version" => "0.08.00d (0.07.01-255-g47407b8; Sep 24 2018 14:49:37)"
        "Config:NCP:DriverName" => "spinel"
        "NCP:HardwareAddress" => [000B57FFFEA77ECE]
        "NCP:Channel" => 11
        "Network:NodeType" => "end-device"
        "Network:Name" => "OpenThreadDemo"
        "Network:XPANID" => 0x1111111122222222
        "Network:PANID" => 0x1234
        "IPv6:LinkLocalAddress" => "fe80::881e:c618:e800:ce3c"
        "IPv6:MeshLocalAddress" => "fd11:1111:1122:0:5fbb:d716:6fa:fbe0"
        "IPv6:MeshLocalPrefix" => "fd11:1111:1122::/64"
        "com.nestlabs.internal:Network:AllowingJoin" => false

"wpanctl get"
Thread:Leader:Weight = 0x40
Thread:NeighborTable: <empty> (7)
get: error: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

cat /var/log/syslog
Sep 24 15:43:43 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:43:43 raspberrypi wpantund[5527]: SendCommand task encountered an error: 16 (0x00000010)
Sep 24 15:43:48 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:43:48 raspberrypi wpantund[5527]: SpinelNCPTaskGetNetworkTopology.cpp:511: Requirement Failed ((ret) == 0)
Sep 24 15:43:48 raspberrypi wpantund[5527]: Getting child/neighbor table failed: 7
Sep 24 15:43:53 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:43:53 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:43:53 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:43:58 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:43:58 raspberrypi wpantund[5527]: SpinelNCPTaskGetNetworkTopology.cpp:511: Requirement Failed ((ret) == 0)
Sep 24 15:43:58 raspberrypi wpantund[5527]: Getting child/neighbor table failed: 7
Sep 24 15:44:03 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:03 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:44:03 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:44:08 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:08 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:44:08 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:44:13 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:13 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:44:13 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:44:18 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:18 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:44:18 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:44:20 raspberrypi wpantund[5527]: SpinelNCPInstance-Protothreads.cpp:232: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:20 raspberrypi wpantund[5527]: NCP is misbehaving or unresponsive
Sep 24 15:44:20 raspberrypi wpantund[5527]: State change: "associated" -> "uninitialized"
Sep 24 15:44:20 raspberrypi wpantund[5527]: Taking interface(s) down. . .
Sep 24 15:44:21 raspberrypi ntpd[805]: Deleting interface #57 wpan0, 2001:db8:8568:b2b5:881e:c618:e800:ce3c#123, interface stats: received=0, sent=0, dropped=0, active_time=124 secs
Sep 24 15:44:21 raspberrypi ntpd[805]: Deleting interface #58 wpan0, fd11:1111:1122:0:5fbb:d716:6fa:fbe0#123, interface stats: received=0, sent=0, dropped=0, active_time=124 secs
Sep 24 15:44:21 raspberrypi ntpd[805]: Deleting interface #59 wpan0, fe80::881e:c618:e800:ce3c%7#123, interface stats: received=0, sent=0, dropped=0, active_time=124 secs
Sep 24 15:44:23 raspberrypi wpantund[5527]: SpinelNCPTask.cpp:86: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:23 raspberrypi wpantund[5527]: SpinelNCPTaskSendCommand.cpp:340: Requirement Failed ((mRetVal) == 0)
Sep 24 15:44:23 raspberrypi wpantund[5527]: SendCommand task encountered an error: 7 (0x00000007)
Sep 24 15:44:25 raspberrypi wpantund[5527]: SpinelNCPInstance-Protothreads.cpp:414: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
Sep 24 15:44:26 raspberrypi wpantund[5527]: Resetting and trying again... (retry 1)

NickNamerr, 2018-09-24:
I have 27 REEDs in the network. After reset, when NeighborTable did not filled yet, it executes command without any issues. Maximum amount of devices in NeighborTable i have seen is 15. I hope it helps.

jwhui, 2018-09-25:
Can you see if the NCP is stuck in a loop? For example, by using JTAG and grabbing a stack trace?

NickNamerr, 2018-09-27:
I shall try to debug it.

abtink, 2018-10-03:
What is the value of `OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE` in your configuration. Can you increase it? 

I have a theory of what's happning. If the the spinel encoded neighbor table frame does not fit in the spinel NCP buffer, the NCP encoder will keep trying to prepare the response and hits the no buffer error and tries again, causing it to remain stuck in a loop.

abtink, 2018-10-05:
That's very likely to be the cause. Can you increase it to at least 1K (preferably 2K)?

jwhui, 2018-11-06:
@NickNamerr, any update on this issue?

closed 2018-12-05 by jwhui