openthread/openthread #3099: CoAP con message - error 28 ResponseTimeout

NickNamerr, 2018-09-26:
Please check attached log of FTD device and screenshot of wireshark from OTBR.

There are clear that server send back CoAP ACK to right destination and port, device log shows that something like ACK received (there only BR and FTD in this network). 
**Device did not handle CoAP ACK.**

I checked same revision POSIX example, everything works fine.

make parameters are :
"make -f examples/Makefile-efr32 SERVICE=1 BORDER_ROUTER=1 COAP=1 UDP_PROXY=1 BOARD=BRD4168A"

OPENTHREAD/20170716-00904-g06bff963-dirty; EFR32; Sep 26 2018 12:38:04

[image]
[coap-error28.log](https://github.com/openthread/openthread/files/2418147/coap-error28.log)

Please advise.

jwhui, 2018-09-26:
@NickNamerr, note that a CoAP ACK is not the same as a CoAP Response. The ACK only acknowledges receipt of the CoAP Request, but does not actually provide the CoAP Response. In the wireshark screenshot, the CoAP ACK is an empty message.

Can you provide more detail on your setup? In particular, enough information so that someone else can reproduce your issue?

NickNamerr, 2018-09-26:
this is log from POSIX example:

> coap start
Coap service started: Done
> 
> 
> coap post fdde:ad00:beef:0:c4ce:c80d:e662:5616 k con 123
Sending coap request: Done
> 
============================================[len =  50]============================
| 69 98 39 34 12 00 44 00 3C 0D 10 00 00 00 01 95| i ? 9 4 ? ? D ? < ? ? ? ? ? ? ?|
| 4F 71 5C C8 65 45 82 62 1C 09 A0 74 F8 C0 47 16| O q \ ? e E ? b ? ? ? t ? ? G ?|
| A0 74 53 CD 7F 48 B0 B2 40 63 39 A9 EB 6F 96 5B| ? t S ? ? H ? ? @ c 9 ? ? o ? [|
| AF 58 .. .. .. .. .. .. .. .. .. .. .. .. .. ..| ? X . . . . . . . . . . . . . .|
-----------------------------------------------------------------------------------

============================================[len =  44]============================
| 69 98 1A 34 12 00 3C 00 44 0D 14 00 00 00 01 55| i ? ? 4 ? ? < ? D ? ? ? ? ? ? U|
| 3F 55 C0 2D A9 A9 7A E5 D0 1C CF DD B1 6A D9 9A| ? U ? - ? ? z ? ? ? ? ? ? j ? ?|
| C9 04 B7 7F A0 DC 8D BC 4E 94 26 5F .. .. .. ..| ? ? ? ? ? ? ? ? N ? & _ . . . .|
-----------------------------------------------------------------------------------
**### Received coap response**

EFR32 does not handle ACK as POSIX. Could you check if I missed any MAKE parameters?

jwhui, 2018-09-26:
What is the CoAP server implementation in your setup? Is it a CoAP server running on OTBR?

NickNamerr, 2018-09-26:
Yes, it node.js coap-server module running on OTBR RPI-based.

jwhui, 2018-09-26:
I suspect the CoAP Server running on OTBR is not configured properly. As mentioned, I do not see any CoAP Response being sent on the wireshark screenshot that you provided. See [RFC 7252 Section 5.2](https://tools.ietf.org/html/rfc7252#section-5.2) for more detail on CoAP Responses.

NickNamerr, 2018-09-26:
Do you mean, that [this](https://github.com/openthread/openthread/blob/2ce47a8e9cf86fc1568f2c2b45e18234193779ce/src/cli/cli_coap.cpp#L311) code does not handle ACK message?

jwhui, 2018-09-26:
As mentioned previously, a CoAP ACK is **_not_** the same as a CoAP Response. The error you are seeing is "ResponseTimeout". The CoAP Server is sending the CoAP ACK, but **_not_** sending the CoAP Response. The CoAP Server must also send a CoAP Response.

NickNamerr, 2018-09-26:
Thanks, this conversation was very useful to figure out what is wrong.

closed 2018-09-27 by jwhui