openthread/openthread #3109: OpenThread Uart

NOA-Ben, 2018-09-30:
I am an openThread development user. I use NRF52840 to run this OpenThread protocol. Because the UART function is used in the project,
              I found that the API interface of OpenThread's UART can only send data, and can not receive serial data in the main function.
             This seems to be a bug, I look at the source code of uart.c, this function does not seem to do anything
             OT_TOOL_WEAK void otPlatUartReceived(const uint8_t *aBuf, uint16_t aBufLength)
             {
                     (void)aBuf;
                    (void)aBufLength;
            }

jwhui, 2018-10-02:
When using the CLI and NCP examples, `otPlatUartReceived()` is implemented by the CLI and NCP modules directly.

https://github.com/openthread/openthread/blob/9f55322d36d113fde31b2fea3c6fd5866d136648/src/cli/cli_uart.cpp#L106

https://github.com/openthread/openthread/blob/9f55322d36d113fde31b2fea3c6fd5866d136648/src/ncp/ncp_uart.cpp#L247

jwhui, 2018-10-11:
@NOA-Ben, were you able to resolve the issue?

NOA-Ben, 2018-10-12:
> @NOA-Ben, were you able to resolve the issue?
I am very sorry to reply to you now. I modified the code of cli_uart.app, blocked the function of cli and recompiled the firmware. Now I can read the serial port data, thank you very much.

wlk05505, 2019-12-28:
@NOA-Ben @jwhui  Hello,NOA-Ben,I want to ask you a question about the source file cli_uart.cpp,how to change it so that I can use Uart Received?This question has troubled me for a long tiime.