openthread/openthread #3137: Suggest to add more EFR32 module series besides EFR32MG12P

swordinrock123, 2018-10-10:
Hello,

I have been using OpenThread in my project for several months, and the EFR32 chip I am using is EFR32MG1B/EFR32MG1P, I can modify the Makefiles and have OpenThread work on my devices.

However, it bothers me that every time when I update OpenThread code from GitHub, I have to do the modifications to Makefile again. Since OpenThread can work on more EFR32 chip series, how about we add them into the Makefile so that we can directly use?

I thought about the solution roughly, but I'm still not sure how to do this.
A:  Use individual folders like TI's chip. such as examples\platforms\efr32mg1, examples\platforms\efr32mg12 ...
B:  Keep the only efr32 folder, add chip series number definition in configure.ac, then use it in Makefile.am for different compile configurations.
C:  Some other ways

Any thoughts about this?

jwhui, 2018-10-11:
It depends how different the EFR32 variants are. In general, I think each directory under `examples/platforms` should cover a chip family. At the same time, if there is a common driver API that can be used across various chips, I could see those being considered as one family.

I think we should try to avoid chip-specific modifications to `configure.ac`.

swordinrock123, 2018-10-12:
@jwhui Thank you for your reply, in my project, I didn't modify the driver of EFR32, which means currently I found no problem that EFR32MG1 share the same driver with EFR32MG12, however, there are some differences between them in Makefile and Makefile.am.

For example, in examples\platforms\efr32\Makefile.am, I need to do the following changes:
1. '-I$(EFR32MG_SDK_SRCDIR)/platform/Device/SiliconLabs/EFR32MG**12P**/Include      \' to
    '-I$(EFR32MG_SDK_SRCDIR)/platform/Device/SiliconLabs/EFR32MG**1B**/Include      \'
2. '@top_builddir@/third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG**12P**/Source/system_efr32mg**12p**.c      \' to '@top_builddir@/third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG**1B**/Source/system_efr32mg**1b**.c      \'
3. '@top_builddir@/third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG12P/Source/GCC/startup_efr32mg**12p**.c \' to '@top_builddir@/third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG1B/Source/GCC/startup_efr32mg**1b**.c \'

Similar modifications are needed in examples\platforms\efr32\Makefile.platform.am.
If I want to use EFR32MG1P, I need to do the modifications again. otherwise it will not work.

As far as I know, if I want to use 'if' command in Makefile.am to achieve conditional compilation, I have to add AM_CONDITIONAL in configure.ac.

Please correct me if I'm miss something, or if you have other good ideas, I'm willing to have a try, thanks!!

jwhui, 2018-10-12:
@swordinrock123, some notes:

1. `examples/Makefile-efr32` already includes [conditionals based on the specific EFR32 board](https://github.com/openthread/openthread/blob/d64b889f5217b2af985c75accc52d2f97858a4c0/examples/Makefile-efr32).
2. `examples/platforms/efr32/Makefile.am` [utilizes the `BOARD` variable when listing include paths](https://github.com/openthread/openthread/blob/d64b889f5217b2af985c75accc52d2f97858a4c0/examples/platforms/efr32/Makefile.am#L57).

Can we follow a similar model?

swordinrock123, 2018-10-15:
@jwhui Hi, I tried the similar model as you told me, but finally I find that it can not work because **automake does not allow it**.

You can check the document of Automake about ["Conditional compilation of sources"](http://www.delorie.com/gnu/docs/automake/automake_39.html)
It says that we can't use configure substitution in _SOURCE

And it also gives two possible ways:
1. [9.1.3.1 Conditional compilation using _LDADD substitutions](http://www.delorie.com/gnu/docs/automake/automake_40.html)
2. [9.1.3.2 Conditional compilation using Automake conditionals](http://www.delorie.com/gnu/docs/automake/automake_41.html)

**It seems that both ways need to modify configure.ac.**

Below is my test steps:
1. Download OpenThread from Github.
2. 'vi examples/platforms/efr32/Makefile.am' and replace EFR32MG12P with **$(CHIPSET)**
3. './bootstrap' and 'make -f examples/Makefile-efr32 BOARD=BRD4161A **CHIPSET=EFR32MG12P**'
4. Then I got error messages:
Makefile:831: ../../../third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG12P/Source/.deps/libopenthread_efr32_a-system_efr32mg12p.Po: No such file or directory
Makefile:832: ../../../third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG12P/Source/GCC/.deps/libopenthread_efr32_a-startup_efr32mg12p.Po: No such file or directory
make[5]: *** No rule to make target '../../../third_party/silabs/gecko_sdk_suite/v2.3/platform/Device/SiliconLabs/EFR32MG12P/Source/GCC/.deps/libopenthread_efr32_a-startup_efr32mg12p.Po'.  Stop.

jwhui, 2018-10-17:
@swordinrock123, what about something similar to what is done in [examples/platforms/efr32/fem-control.c](https://github.com/openthread/openthread/blob/a56666a6d0d6e1edf1f4bcd242bc0fed63ce6a76/examples/platforms/efr32/fem-control.c)?

swordinrock123, 2018-11-02:
@jwhui Hello, I followed your suggestion and it can work. But there still remains another problem.

In [https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/Makefile.am#L57](url), it requires the board config path such as EFR32MG12_BRD4161A, as we know, these boards are all from SiLabs, what if someone is using board from other vendors with EFR32 as radio chipset?

After research, I know if I define -DHAL_CONFIG=1, then I can remove this line, because 'silabs/gecko_sdk_suite/v2.3/hardware/kit/common/bsp/bsp.h' can use different header file according to HAL_CONFIG:
```
#if defined(HAL_CONFIG)
#include "bsphalconfig.h"
#else
#include "bspconfig.h"
#endif
```
bsphalconfig.h is in common folder 'gecko_sdk_suite\v2.3\hardware\kit\common\halconfig'
bspconfig.h is in individual folder such as 'gecko_sdk_suite\v2.3\hardware\kit\EFR32MG12_BRD4161A'

So I wonder if it is necessary to bind openthread to silabs boards only? or we just use common bsphalconfig.h for all boards that uses EFR32 as radio chipset?

jwhui, 2018-11-02:
I think it makes sense to allow custom boards. One possibility is to add a "custom" option for the BOARD specifier. When "custom" is specified, the board-specific include path is not added. Thoughts?

jwhui, 2018-12-05:
@swordinrock123, an updates on this issue?

jwhui, 2019-04-23:
Closing stale issue.

closed 2019-04-23 by jwhui