openthread/openthread #3143: Coap Error receiving coap response message (Error 28: ResponseTimeout)

kpyush, 2018-10-11:
**Describe the bug**
Node is not able to receive coap response while we are sending packet on coap server.

**To Reproduce**
Hardware: NRF52840DK
OTBR on raspberry pi

Build steps:
    1. coap resource data
    2. coap start
     3. coap put 64:ff9b::68d3:64ed data con message

**Expected behavior**
It should receive proper response which being send by server.

**Console/log output**
Attached.
[image]
[image]

**Additional context**
Similar issue #2733

kpyush, 2018-10-12:
We are receiving timeout even in piggybacked coap server. Worse thing is sniffer is getting the response and ack packets properly but device still sends duplicate packets till the maximum limit after that it throws response error.

In this log client is sending coap get packet on a ip address (2001:db8:1:ffff::c0a8:a64) with **test** resource defined.

cli-ftd example is being used as reference. (with coap enabled)

[image]

LuDuda, 2018-10-12:
From attached logs i see that your second problem is the same as reported in #2733 which you mentioned before.

Server responds from different IPv6 address. Check first reply from @jwhui in the #2733.

kpyush, 2018-10-13:
@LuDuda do you have any idea how to fix this #2733, If not then where should I look into OTBR or in CLI-FTD APP in order to resolve the issue.

LuDuda, 2018-10-13:
The problem seems to be located in the CoAP Server implementation in the OTBR. Which implementation do you use?

I see one good solution and two workarounds:

1. CoAP Server should respond with the source address equal to the destination address of the request. This should be ensured in the CoAP Server implementation or its configuration (e.g. maybe you can bind to specific address). If its not, i suggest to fix it there or change CoAP Server implementation.
2. I see that destination address in CoAP Requests is `2001:db8:1:ffff::c0a8:a64` while response comes from `2001:db8:1:ffff::c0a8:ff01`. Maybe you can just try to make request to `2001:db8:1:ffff::c0a8:ff01`.
3. Not recommended - only for testing purpose. You can remove three first checks from this [line](https://github.com/openthread/openthread/blob/master/src/core/coap/coap.cpp#L489) and see if behaviour becomes expected.

kpyush, 2018-10-15:
Coap server library: [Coapython](https://github.com/Tanganelli/CoAPthon)
The ip address **2001:db8:1:ffff::c0a8:ff01** is of NAT64. And **2001:db8:1:ffff::c0a8:a64** is of wlan0. The CoAP server is running in private network. When I tried to send CoAP message at destination address **2001:db8:1:ffff::c0a8:ff01**, then the packet is not able to reach my local server running on RPi. 

I think we have to enable routing between NAT64 and wlan0. Can you please help in this issue.

LuDuda, 2018-10-16:
Great to hear that!