openthread/openthread #3185: Makefile nRF52840 not working on Ubuntu 18.04!

nikiizvorski, 2018-10-19:
**Describe the bug**
Makefile nRF52840 not working on Ubuntu 18.04 doesn't work and create a build the rest Makefiles do.

**To Reproduce**
Information to reproduce the behavior, including:
1. pull latest commit
2. Execute
$ cd ~/openthread
$ make -f examples/Makefile-nrf52840 clean
$ make -f examples/Makefile-nrf52840 JOINER=1

**Expected behavior**
Just build the build that is required from it.

**Console/log output**
make -f examples/Makefile-nrf52840 JOINER=1
  MKDIR    output
  MKDIR    build/nrf52840
  CONFIG   nrf52840...
(cd build/nrf52840 && /home/viblegrid/openthread/examples/../configure INSTALL="/usr/bin/install -p" CPP="arm-none-eabi-cpp" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++" CCAS="arm-none-eabi-gcc" OBJC="cc" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" STRIP="arm-none-eabi-strip" CPPFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs " CFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs " CXXFLAGS=" -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -fno-exceptions -fno-rtti " CCASFLAGS="  -x assembler-with-cpp " LDFLAGS="-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='\"openthread-core-nrf52840-config.h\"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections " --host=arm-none-eabi --prefix=/ --exec-prefix=/nrf52840 --enable-cli --enable-diag --enable-ftd --enable-mtd --enable-ncp --enable-radio-only --enable-linker-map --with-examples=nrf52840 MBEDTLS_CPPFLAGS="-DMBEDTLS_CONFIG_FILE='\"mbedtls-config.h\"' -DMBEDTLS_USER_CONFIG_FILE='\"nrf52840-mbedtls-config.h\"' -I/home/viblegrid/openthread/examples/../third_party/mbedtls -I/home/viblegrid/openthread/examples/../third_party/mbedtls/repo/include -I/home/viblegrid/openthread/examples/../third_party/mbedtls/repo/include/mbedtls -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/libraries/crypto -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/libraries/nrf_cc310/include -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/nrfx/mdk -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/cmsis -DNRF52840_XXAA"  --with-ncp-bus=uart --enable-joiner)
checking build system type... x86_64-unknown-linux-gnu
checking host system type... arm-none-eabi
checking filtered build system type... x86_64-unknown-linux-gnu
checking filtered host system type... arm-none-eabi
checking for a BSD-compatible install... /usr/bin/install -p
checking whether build environment is sane... yes
checking for arm-none-eabi-strip... arm-none-eabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether to disable executable checking... no
checking for style of include used by make... GNU
checking for arm-none-eabi-gcc... arm-none-eabi-gcc
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-none-eabi-gcc accepts -g... yes
checking for arm-none-eabi-gcc option to accept ISO C89... none needed
checking whether arm-none-eabi-gcc understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking how to run the C preprocessor... arm-none-eabi-cpp
checking for arm-none-eabi-gcc... (cached) arm-none-eabi-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether arm-none-eabi-gcc accepts -g... (cached) yes
checking for arm-none-eabi-gcc option to accept ISO C89... (cached) none needed
checking whether arm-none-eabi-gcc understands -c and -o together... (cached) yes
checking dependency style of arm-none-eabi-gcc... (cached) gcc3
checking whether arm-none-eabi-gcc and cc understand -c and -o together... yes
checking whether we are using the GNU C++ compiler... yes
checking whether arm-none-eabi-g++ accepts -g... yes
checking dependency style of arm-none-eabi-g++... gcc3
checking how to run the C++ preprocessor... arm-none-eabi-g++ -E
checking whether we are using the GNU C++ compiler... (cached) yes
checking whether arm-none-eabi-g++ accepts -g... (cached) yes
checking dependency style of arm-none-eabi-g++... (cached) gcc3
checking whether arm-none-eabi-g++ understands -c and -o together... yes
checking dependency style of arm-none-eabi-gcc... gcc3
checking for arm-none-eabi-ar... arm-none-eabi-ar
checking for arm-none-eabi-ranlib... arm-none-eabi-ranlib
checking for arm-none-eabi-objcopy... arm-none-eabi-objcopy
checking for arm-none-eabi-strip... (cached) arm-none-eabi-strip
checking whether ln -s works... yes
checking for cmp... /usr/bin/cmp
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
[... 116 lines cut ...]
checking whether to enable examples... 
checking whether to build tools... yes
checking for doxygen... no
checking for dot... no
checking whether to build documentation... no
configure: checking required package dependencies
checking for stdbool.h that conforms to C99... yes
checking for _Bool... yes
checking for ANSI C header files... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking for strlcpy... configure: error: link tests are not allowed after AC_NO_EXECUTABLES
examples/Makefile-nrf52840:282: recipe for target 'build/nrf52840/config.status' failed
make: *** [build/nrf52840/config.status] Error 1

**Additional context**
We have try to build the other Makefiles seem to work fine. Let me know how to help to advance this problem quicker. Thanks

OS: Ubuntu 18.04, Screen version 4.06.02 (GNU) 23-Oct-17, gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
, chmod -R -w "${HOME}"/opt/gnu-mcu-eclipse/arm-none-eabi-gcc/7.3.1-1.1-20180724-0637/bin/arm-none-eabi-gcc --version
chmod (GNU coreutils) 8.28

GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.

nikiizvorski, 2018-10-19:
Hey guys a solution to this would be if someone can build a nRF52840 ot-cli.hex for us on the latest commit and pass it to us. Thanks a lot!

bukepo, 2018-10-19:
I have tested on 18.04, it works as expected. Could you please clean your worktree first?
```
git clean -dfx
```

nikiizvorski, 2018-10-19:
Hey @bukepo just did that same result i have posted to tools i am using. If its working for you can you pass a cli hex from the latest so we can have a solution for now. Thanks

bukepo, 2018-10-19:
I verified, the tool versions are the same. Could you please attach your `build/nrf52840/config.log`?

Besides, you may also try reinstalling the toolchain though `apt-get`.
```
./script/bootstrap
```

nikiizvorski, 2018-10-19:
Output:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by OPENTHREAD configure 20170716-00988-g71d4ad3e, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /home/viblegrid/openthread/examples/../configure INSTALL=/usr/bin/install -p CPP=arm-none-eabi-cpp CC=arm-none-eabi-gcc CXX=arm-none-eabi-g++ CCAS=arm-none-eabi-gcc OBJC=cc OBJCXX= AR=arm-none-eabi-ar RANLIB=arm-none-eabi-ranlib NM=arm-none-eabi-nm STRIP=arm-none-eabi-strip CPPFLAGS=-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-nrf52840-config.h"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs  CFLAGS= -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-nrf52840-config.h"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs  CXXFLAGS= -fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-nrf52840-config.h"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -fno-exceptions -fno-rtti  CCASFLAGS=  -x assembler-with-cpp  LDFLAGS=-fdata-sections -ffunction-sections -Os -g -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE='"openthread-core-nrf52840-config.h"' -I/home/viblegrid/openthread/examples/../examples/platforms/nrf52840/  -DOPENTHREAD_CONFIG_LOG_OUTPUT=OPENTHREAD_CONFIG_LOG_OUTPUT_PLATFORM_DEFINED -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -specs=nano.specs -specs=nosys.specs -Wl,--gc-sections  --host=arm-none-eabi --prefix=/ --exec-prefix=/nrf52840 --enable-cli --enable-diag --enable-ftd --enable-mtd --enable-ncp --enable-radio-only --enable-linker-map --with-examples=nrf52840 MBEDTLS_CPPFLAGS=-DMBEDTLS_CONFIG_FILE='"mbedtls-config.h"' -DMBEDTLS_USER_CONFIG_FILE='"nrf52840-mbedtls-config.h"' -I/home/viblegrid/openthread/examples/../third_party/mbedtls -I/home/viblegrid/openthread/examples/../third_party/mbedtls/repo/include -I/home/viblegrid/openthread/examples/../third_party/mbedtls/repo/include/mbedtls -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/libraries/crypto -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/libraries/nrf_cc310/include -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/nrfx/mdk -I/home/viblegrid/openthread/examples/../third_party/NordicSemiconductor/cmsis -DNRF52840_XXAA --with-ncp-bus=uart --enable-commissioner --enable-joiner

## --------- ##
## Platform. ##
## --------- ##

hostname = viblegrid-Macmini
uname -m = x86_64
uname -r = 4.15.0-36-generic
uname -s = Linux
uname -v = #39-Ubuntu SMP Mon Sep 24 16:19:09 UTC 2018

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games
PATH: /snap/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2968: checking build system type
configure:2982: result: x86_64-unknown-linux-gnu
configure:3002: checking host system type
configure:3015: result: arm-none-eabi
configure:3045: checking filtered build system type
configure:3053: result: x86_64-unknown-linux-gnu
configure:3060: checking filtered host system type
configure:3068: result: arm-none-eabi
configure:3098: checking for a BSD-compatible install
configure:3166: result: /usr/bin/install -p
configure:3177: checking whether build environment is sane
configure:3232: result: yes
configure:3291: checking for arm-none-eabi-strip
configure:3318: result: arm-none-eabi-strip
configure:3383: checking for a thread-safe mkdir -p
configure:3422: result: /bin/mkdir -p
configure:3429: checking for gawk
configure:3445: found /usr/bin/gawk
configure:3456: result: gawk
configure:3467: checking whether make sets $(MAKE)
configure:3489: result: yes
configure:3518: checking whether make supports nested variables
configure:3535: result: yes
configure:3614: checking how to create a pax tar archive
configure:3625: tar --version
[... 1455 lines cut ...]
#define OPENTHREAD_ENABLE_CHANNEL_MONITOR 0
#define OPENTHREAD_ENABLE_CHANNEL_MANAGER 0
#define OPENTHREAD_ENABLE_MAC_FILTER 0
#define OPENTHREAD_ENABLE_DIAG 1
#define OPENTHREAD_ENABLE_LEGACY 0
#define OPENTHREAD_ENABLE_NCP_VENDOR_HOOK 0
#define OPENTHREAD_ENABLE_NCP_SPINEL_ENCRYPTER 0
#define OPENTHREAD_ENABLE_CHILD_SUPERVISION 0
#define OPENTHREAD_ENABLE_CERT_LOG 0
#define OPENTHREAD_ENABLE_DHCP6_CLIENT 0
#define OPENTHREAD_ENABLE_DHCP6_SERVER 0
#define OPENTHREAD_ENABLE_DNS_CLIENT 0
#define OPENTHREAD_ENABLE_APPLICATION_COAP 0
#define OPENTHREAD_ENABLE_RAW_LINK_API 0
#define OPENTHREAD_ENABLE_BORDER_ROUTER 0
#define OPENTHREAD_ENABLE_SERVICE 0
#define OPENTHREAD_EXAMPLES_NRF52840 1
#define OPENTHREAD_ENABLE_EXAMPLES nrf52840
#define HAVE__BOOL 1
#define HAVE_STDBOOL_H 1
#define STDC_HEADERS 1
#define HAVE_STDINT_H 1
#define HAVE_STRING_H 1

configure: exit 1

nikiizvorski, 2018-10-19:
Thanks a lot for the hex file!

bukepo, 2018-10-19:
It seems to be caused by the tool chain. Could you please try remove the toolchain you installed and just run the following command to install toolchain?
`./script/bootstrap`

nikiizvorski, 2018-10-19:
Can you suggest a command for that? Thanks

bukepo, 2018-10-19:
what command?

KaSroka, 2018-10-22:
I can confirm that it's toolchain related - gcc-arm-none-eabi that comes with Ubuntu 18.04 LTS has problem with ld. Try to uninstall it with `sudo apt remove gcc-arm-none-eabi` and install it from here https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads (I'm using version 6-2017-q2-update Linux 64bit and it works perfectly fine)

jwhui, 2018-10-22:
Yes, every instance of:
```
configure: error: link tests are not allowed after AC_NO_EXECUTABLES
```
that I've seen points to an issue with the GNU ARM Embedded Toolchain and can be solved by reinstalling the toolchain.

nikiizvorski, 2018-10-25:
I can confirm thats the solution and it works fine.

nikiizvorski, 2018-10-25:
Can anyone build an hex or bin for cc2652? Would really appreciate it. Thanks

jwhui, 2018-11-06:
@nikiizvorski, CC2652 binaries: [ot-ncp-ftd-g38bd2569-cc2652.tar.gz](https://github.com/openthread/openthread/files/2554960/ot-ncp-ftd-g38bd2569-cc2652.tar.gz)

Closing, since the original issue has been resolved.

closed 2018-11-06 by jwhui