openthread/openthread #3219: [efr32] Project doesn't compile when OPENTHREAD_CONFIG_ENABLE_TIME_SYNC is enabled

josephn123, 2018-10-26:
**Describe the bug**
When enabling `OPENTHREAD_CONFIG_ENABLE_TIME_SYNC` on efr32 config, the project will not compile because of the missing includes:
```
#include <openthread/instance.h>
#include <stdint.h>
```

**To Reproduce**
Change `openthread-core-efr32-config.h` `OPENTHREAD_CONFIG_ENABLE_TIME_SYNC` to 1. 

**Expected behavior**
Project compiles

**Console/log output**
Snippet:
```
In file included from /home/joseph/repos/openthread/examples/../src/core/thread/time_sync_service.hpp:40:0,
                 from /home/joseph/repos/openthread/examples/../src/core/thread/time_sync_service.cpp:38:
/home/joseph/repos/openthread/examples/../include/openthread/network_time.h:79:38: error: 'otInstance' was not declared in this scope
 otNetworkTimeStatus otNetworkTimeGet(otInstance *aInstance, uint64_t &aNetworkTime);

/home/joseph/repos/openthread/examples/../include/openthread/network_time.h:79:61: error: 'uint64_t' was not declared in this scope
 otNetworkTimeStatus otNetworkTimeGet(otInstance *aInstance, uint64_t &aNetworkTime);
```

**Additional context**
Willing to fix this, however I am not sure where is best to put the required includes. I can't work out how these files are included for other platforms, i.e. nrf52840.

I am aware time sync doesn't work for the efr32. Some context: I have been working on a non-compliant hack (placing header IEs in the payload) to get it working for a very specific use case on the efr32.

jwhui, 2018-10-26:
What commit-id are you using? I believe the build errors noted above were resolved by #3167.

josephn123, 2018-10-30:
Build issue has been fixed, thank you. Closing this issue.

closed 2018-10-30 by josephn123