openthread/openthread #3233: otLinkActiveScan() returns a non null-terminated network name

sergeuz, 2018-10-30:
**Describe the bug**

`otLinkActiveScan()` doesn't null-terminate a network name if it has the maximum length of 16 characters.

**To Reproduce**

Set up a network with a name containing 16 characters and perform an active scan.

**Additional context**

The issue is caused by an incorrect size argument passed to `memcpy()` in the `Mac::ConvertBeaconToActiveScanResult()` method (the source `BeaconPayload::mNetworkName` string is not null-terminated): https://github.com/openthread/openthread/blob/9d585edc46999b82b30b8cf3e88901f8b271ab63/src/core/mac/mac.cpp#L312

jwhui, 2018-10-30:
@sergeuz, thanks for raising this issue! Are you willing to submit a PR with the fix?

sergeuz, 2018-10-30:
@jwhui Sure. I won't be able to make it today though, but will definitely submit it tomorrow if that's ok

jwhui, 2018-10-30:
@sergeuz, of course, no problem!

jwhui, 2018-11-06:
Resolved by #3246.

closed 2018-11-06 by jwhui