openthread/openthread #3236: nRF52840 - Board stopped working with Android Things

nikiizvorski, 2018-10-30:
**Describe the bug**
We have been buying nRF5280 boards and flashing them from open thread website in the first couple tests the devices work properly and connect to each other. We have tried to connect one to one and the devices do connect to each other. After some point we decided to connect to two or tree devices at the same time. In a short period of time we started noticing that some of the chips doesn't want to connect anymore and don't want to work properly. Only some of them are connecting instantly the rest are not even connecting. We were interested to understand if anyone other have this issue and what would be the issue. The device keeps saying no network and after some point its connecting attaching but can't find the network and keeps re-try. And basically after that state is not usable.

**To Reproduce**
1. Last Android Things
2. Last build from OpenThread website
3. Flash and connect
4. Use straight away from the demo

**Expected behavior**
Just to understand where the issue is coming since we want to buy more kits and we already bought around 5-6 so far. 

**Console/log output**

2018-10-30 10:08:05.360 1648-1648/? I/wpantund: [-NCP-]: NCP was reset (STATUS_RESET_POWER_ON, 112)
2018-10-30 10:08:05.360 1648-1648/? I/wpantund: State change: "offline:commissioned" -> "uninitialized"
2018-10-30 10:08:05.381 774-774/com.google.android.gms.persistent W/ChimeraUtils: Module com.google.android.gms.nearby has empty metadata display_name_string_id
2018-10-30 10:08:05.383 981-1404/com.google.android.gms I/Authzen: [PermitStore] Getting all permits...
2018-10-30 10:08:05.393 774-774/com.google.android.gms.persistent W/ChimeraUtils: Module com.google.android.gms.nearby has empty metadata display_name_string_id
2018-10-30 10:08:05.442 1592-1592/com.example.things.lowpan.transmitter I/TransmitterActivity: connect requested
2018-10-30 10:08:05.443 1592-1592/com.example.things.lowpan.transmitter I/TransmitterActivity: connect requested2
2018-10-30 10:08:05.443 1592-1647/com.example.things.lowpan.transmitter I/TransmitterActivity: doInBackground: Connecting...
2018-10-30 10:08:05.444 1592-1647/com.example.things.lowpan.transmitter I/TransmitterActivity: doInBackground: Create socket to /fdfc:6b43:b081:0:f68b:bb45:98be:16af port 23456
2018-10-30 10:08:05.450 177-1527/? W/StreamHAL: Error from HAL stream in function get_presentation_position: Operation not permitted
2018-10-30 10:08:05.452 1592-1647/com.example.things.lowpan.transmitter E/TransmitterActivity: Connection attempt failed
    java.net.SocketException: Binding socket to network 103 failed: ENONET (Machine is not on the network)
        at android.net.Network.bindSocket(Network.java:354)
        at android.net.Network.bindSocket(Network.java:323)
        at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:184)
        at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:171)
        at com.example.things.lowpan.transmitter.TransmitterActivity.startThreadChannel(TransmitterActivity.java:675)
        at com.example.things.lowpan.transmitter.TransmitterActivity.access$1300(TransmitterActivity.java:63)
        at com.example.things.lowpan.transmitter.TransmitterActivity$7.run(TransmitterActivity.java:589)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: android.system.ErrnoException: Binding socket to network 103 failed: ENONET (Machine is not on the network)
        at android.net.Network.bindSocket(Network.java:354) 
        at android.net.Network.bindSocket(Network.java:323) 
        at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:184) 
        at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:171) 
        at com.example.things.lowpan.transmitter.TransmitterActivity.startThreadChannel(TransmitterActivity.java:675) 
        at com.example.things.lowpan.transmitter.TransmitterActivity.access$1300(TransmitterActivity.java:63) 
        at com.example.things.lowpan.transmitter.TransmitterActivity$7.run(TransmitterActivity.java:589) 
        at android.os.Handler.handleCallback(Handler.java:790) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
2018-10-30 10:08:05.452 1592-1647/com.example.things.lowpan.transmitter W/System.err: java.net.SocketException: Binding socket to network 103 failed: ENONET (Machine is not on the network)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.net.Network.bindSocket(Network.java:354)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.net.Network.bindSocket(Network.java:323)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:184)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.net.Network$NetworkBoundSocketFactory.createSocket(Network.java:171)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at com.example.things.lowpan.transmitter.TransmitterActivity.startThreadChannel(TransmitterActivity.java:675)
2018-10-30 10:08:05.453 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at com.example.things.lowpan.transmitter.TransmitterActivity.access$1300(TransmitterActivity.java:63)
2018-10-30 10:08:05.456 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at com.example.things.lowpan.transmitter.TransmitterActivity$7.run(TransmitterActivity.java:589)
2018-10-30 10:08:05.456 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.os.Handler.handleCallback(Handler.java:790)
2018-10-30 10:08:05.456 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
2018-10-30 10:08:05.456 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.os.Looper.loop(Looper.java:164)
2018-10-30 10:08:05.456 1592-1647/com.example.things.lowpan.transmitter W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)
2018-10-30 10:08:05.457 1592-1647/com.example.things.lowpan.transmitter W/System.err: Caused by: android.system.ErrnoException: Binding socket to network 103 failed: ENONET (Machine is not on the network)
2018-10-30 10:08:05.458 1592-1647/com.example.things.lowpan.transmitter W/System.err: 	... 11 more
2018-10-30 10:08:05.458 1592-1647/com.example.things.lowpan.transmitter I/TransmitterActivity: doInBackground: Connecting...
2018-10-30 10:08:05.459 1592-1647/com.example.things.lowpan.transmitter I/TransmitterActivity: doInBackground: Create socket to /fdfc:6b43:b081:0:623f:e72f:295b:c382 port 23456
2018-10-30 10:08:05.460 177-1527/? W/StreamHAL: Error from HAL stream in function get_presentation_position: Operation not permitted
2018-10-30 10:08:05.468 1592-1647/com.example.things.lowpan.transmitter E/TransmitterActivity: Connection attempt failed
    java.net.SocketException: Binding socket to network 103 failed: ENONET (Machine is not on the network)
        at android.net.Network.bindSocket(Network.java:354)
        at android.net.Network.bindSocket(Network.java:323)
[... 350 lines cut ...]
    
    I2C
    I2C1
    
    SPI
    SPI0.0, SPI0.1
    
    GPIO
    BCM10, BCM11, BCM12, BCM13, BCM14, BCM15, BCM16, BCM17, BCM18, BCM19, BCM2, BCM20, BCM21, BCM22, BCM23, BCM24, BCM25, BCM26, BCM27, BCM3, BCM4, BCM5, BCM6, BCM7, BCM8, BCM9
2018-10-30 10:08:11.280 1662-1683/com.android.bluetooth D/BluetoothAdapterProperties: Discoverable Timeout:120
2018-10-30 10:08:11.286 1662-1683/com.android.bluetooth D/BluetoothAdapterProperties: Scan Mode:21
2018-10-30 10:08:11.290 173-235/? D/gralloc: Registering a buffer in the process that created it. This may cause memory ordering problems.
2018-10-30 10:08:11.303 298-315/system_process W/system_server: kill(-1592, 9) failed: No such process
2018-10-30 10:08:11.309 298-309/system_process E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 108)
[... cut ...]
2018-10-30 10:08:11.333 173-837/? D/gralloc: Registering a buffer in the process that created it. This may cause memory ordering problems.
2018-10-30 10:08:11.333 298-309/system_process W/LowpanServiceImpl: LoWPAN interface `wpan1` (android.net.lowpan.ILowpanInterface$Stub$Proxy@f5770f2) died.
2018-10-30 10:08:11.333 298-309/system_process I/LowpanInterfaceTracker: unregister()
2018-10-30 10:08:11.333 298-309/system_process D/LowpanInterfaceTracker: quit:
2018-10-30 10:08:11.334 298-309/system_process I/LowpanServiceImpl: Removed LoWPAN interface `wpan1` (android.net.lowpan.ILowpanInterface$Stub$Proxy@f5770f2)

nikiizvorski, 2018-10-30:
And the second part that we trigger after:

2018-10-30 10:59:29.319 1880-1880/com.example.things.lowpan.transmitter E/TransmitterActivity: Could not provision network
    com.google.android.things.lowpan.LowpanException: android.net.lowpan.JoinFailedException: android.os.ServiceSpecificException: JoinFailedUnknown (code 12)
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:516)
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:508)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: android.net.lowpan.JoinFailedException: android.os.ServiceSpecificException: JoinFailedUnknown (code 12)
        at android.net.lowpan.LowpanInterface.join(LowpanInterface.java:256)
        at com.google.android.things.lowpan.LowpanInterface$2.run(LowpanInterface.java:639)
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:513)
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:508) 
        at android.os.AsyncTask$2.call(AsyncTask.java:333) 
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) 
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
        at java.lang.Thread.run(Thread.java:764) 
     Caused by: android.os.ServiceSpecificException: JoinFailedUnknown (code 12)
        at android.os.Parcel.readException(Parcel.java:2019)
        at android.os.Parcel.readException(Parcel.java:1951)
        at android.net.lowpan.ILowpanInterface$Stub$Proxy.join(ILowpanInterface.java:789)
        at android.net.lowpan.LowpanInterface.join(LowpanInterface.java:250)
        at com.google.android.things.lowpan.LowpanInterface$2.run(LowpanInterface.java:639) 
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:513) 
        at com.google.android.things.lowpan.LowpanInterface$AsyncCallGlue.doInBackground(LowpanInterface.java:508) 
        at android.os.AsyncTask$2.call(AsyncTask.java:333) 
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) 
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
        at java.lang.Thread.run(Thread.java:764) 
2018-10-30 10:59:29.421 1918-1918/? I/wpantund: State change: "uninitialized" -> "offline"
2018-10-30 10:59:29.426 1918-1918/? I/wpantund: NCP is running "OPENTHREAD/20170716-00798-g0c07c53c; NRF52840; Aug 13 2018 14:22:45"
2018-10-30 10:59:29.426 1918-1918/? I/wpantund: Driver is running "0.08.00d (0.07.01-130-g2a5f9cd/0.07.01-130-g2a5f9cd-dirty; Sep 17 2018 23:20:19)"
2018-10-30 10:59:29.472 1918-1918/? I/wpantund: Network is not joinable
2018-10-30 10:59:29.490 1918-1918/? I/wpantund: Resetting interface(s). . .
2018-10-30 10:59:29.491 297-1496/system_process D/LowpanInterfaceTracker: handleMessage: E msg.what=544773
2018-10-30 10:59:29.491 297-1496/system_process D/LowpanInterfaceTracker: processMsg: OfflineState
2018-10-30 10:59:29.491 297-1496/system_process D/LowpanInterfaceTracker: processMsg: NormalState
2018-10-30 10:59:29.491 297-1496/system_process D/LowpanInterfaceTracker: processMsg: DefaultState
2018-10-30 10:59:29.491 297-1496/system_process D/LowpanInterfaceTracker: handleMessage: X
2018-10-30 10:59:29.491 1918-1918/? I/wpantund: Finished initializing NCP
2018-10-30 10:59:36.099 805-805/com.google.android.gms.persistent I/BeaconBle: 'L' hardware scan: scan stopped, no clients

jwhui, 2018-10-31:
Unfortunately, we do not have anyone actively supporting OpenThread on Android Things at this time - of course, contributions are welcome!

Otherwise, is it possible for you to use Raspbian on RPi 3b with [wpantund](https://github.com/openthread/wpantund)? Alternatively, you could try [OpenThread Border Router](https://openthread.io/guides/border-router).

closed 2018-11-06 by jwhui