openthread/openthread #3271: [nrf52840] support DFU Trigger Library (USB)?

bukepo, 2018-11-05:
**Is your feature request related to a problem? Please describe.**
Even build with BOOTLOADER=1, the DFU trigger is not supported.

**Describe the solution you'd like**
Include DFU trigger feature.

**Describe alternatives you've considered**
None.

**Additional context**
I tried to add this feature, but I found there are too many dependencies that are not included in `third_party/NordicSemiconductor/`, so I think it will be more proper to add this feature by Nordic team.

LuDuda, 2018-11-06:
Thanks @bukepo for reporting this issue.

We will take care of that. Note that when it comes to PC side, we currently have support for it only in nRF Connect, is that okay, or you expecting also some sample Linux application?

LuDuda, 2018-11-06:
That is my current understanding. I will consult with the nrfutil team and let you know asap.

librasungirl, 2018-11-07:
@LuDuda , are there any plan for it?

LuDuda, 2018-11-12:
@librasungirl @bukepo our app team hasn't made plans to add DFU trigger to nrfutil (they need to estimate the effort).

There is also the possibility of using pc-nrf-dfu-js (https://github.com/NordicSemiconductor/pc-nrf-dfu-js), or the nrf-device-setup (https://github.com/NordicSemiconductor/nrf-device-setup-js).
They do not have a CLI like nrfutil, but are callable from node.js script. Our app team uses them for programming in nRF Connect for Desktop.

Let me know if that is good enough for you, and if this feature request is still valid (embedded side)? Else contact me directly by e-mail.

LuDuda, 2019-01-17:
Discussed F2F. The DFU Trigger support in nrfutil will be available within 4-6 weeks.

Until it happens [nrf-device-setup-js](https://github.com/NordicSemiconductor/nrf-device-setup-js) component can be used instead.

Below instruction for the attached script ([dfu-trigger.zip](https://github.com/openthread/openthread/files/2768240/dfu-trigger.zip)):

Prerequisite: 
nodejs + npm installed (node 8.x 32 bit, e.g. https://nodejs.org/download/release/v8.15.0/)

Setup:
Create an empty folder somewhere and CD into it
Install the module nrf-device-setup
`$ npm install nrf-device-setup`

Copy the attached file into the folder.

Execute the script with:
`$ node dfu-trigger -s <serialnumber>`

You can also use interactive mode (-i instead -s) to retrieve serial numbers of attached devices (USB serial number similar to JLink number is permanent).

@jwhui I think we can close this issue, since both firmware and software are in place.

jwhui, 2019-01-17:
@LuDuda, thanks!

closed 2019-01-17 by jwhui