openthread/openthread #3300: Not able compile OpenThread on CC1352

bahubalioffice67, 2018-11-14:
I am trying to compile the openthread  for cc1352. I am following steps given below : 
$ ./script/bootstrap
$ ./bootstrap
$ make -f examples/Makefile-cc1352

 I am getting following error while compilation
~/openthread$ make -f examples/Makefile-cc1352
  BUILD    cc1352
make -j10 -C build/cc1352 --no-print-directory all
  CHECK    .local-version
make --no-print-directory all-recursive
Making all in include
make  all-recursive
Making all in openthread
Making all in platform
make[6]: Nothing to be done for 'all'.
make[6]: Nothing to be done for 'all-am'.
Making all in third_party
Making all in mbedtls
make[4]: Nothing to be done for 'all'.
make[4]: Nothing to be done for 'all-am'.
Making all in src
Making all in core
make[4]: Nothing to be done for 'all'.
Making all in cli
make[4]: Nothing to be done for 'all'.
Making all in ncp
make[4]: Nothing to be done for 'all'.
Making all in diag
make[4]: Nothing to be done for 'all'.
make[4]: Nothing to be done for 'all-am'.
Making all in examples
Making all in platforms
Making all in utils
make[5]: Nothing to be done for 'all'.
Making all in cc1352
make[5]: Nothing to be done for 'all'.
make[5]: Nothing to be done for 'all-am'.
Making all in apps
Making all in cli
  CCLD     ot-cli-ftd
  CCLD     ot-cli-mtd
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-assert.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-assert.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-fprintf.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-fprintf.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-impure.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-impure.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-isprint.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-isprint.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memcmp.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memcmp.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memcpy-stub.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memcpy-stub.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memmove.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memmove.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memset.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-memset.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-nano-vfprintf.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-nano-vfprintf.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-nano-vfprintf_i.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-nano-vfprintf_i.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-snprintf.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-snprintf.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strchr.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strchr.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strcpy.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strcpy.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-mtd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strlen-stub.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libg_nano.a(lib_a-strlen-stub.o)
[... 178 lines cut ...]
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-ftd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(kill.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(kill.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-ftd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(lseek.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(lseek.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-ftd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(read.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(read.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ot-cli-ftd uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libnosys.a(sbrk.o) does not
[... cut ...]
Makefile:1174: recipe for target 'ot-cli-ftd' failed
make[5]: *** [ot-cli-ftd] Error 1
Makefile:486: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
Makefile:489: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:657: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
Makefile:583: recipe for target 'all' failed
make[1]: *** [all] Error 2
examples/Makefile-cc1352:261: recipe for target 'do-build-cortex-m4f' failed
make: *** [do-build-cortex-m4f] Error 2

jwhui, 2018-11-14:
@bahubalioffice67, can you provide more information about your build system (e.g. operating system, distribution, etc.).

One thing to try is uninstalling your arm-gcc toolchain then downloading and installing the latest directly from: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

bahubalioffice67, 2018-11-14:
OS:Ubuntu(18.04.1 LTS 64bit) running on virtualBox

jwhui, 2018-11-14:
Makefile-cc1352 [configures](https://github.com/openthread/openthread/blob/33b0ee93308345316ca7a8a1ddc8aa86ba42983e/examples/Makefile-cc1352#L253) the compiler to use hard-float ABI. However, it appears that the linker is trying to link against a library built using soft-float ABI (i.e. `arm-none-eabi/lib/libg_nano.a`).

Instead, it should be trying to link against the library build using hard-float ABI (i.e. `rm-none-eabi/lib/hard/libg_nano.a`). Do you have the hard-float ABI version installed?

As mentioned above, you can try reinstalling your arm-gcc toolchain.

bahubalioffice67, 2018-11-15:
Thanks @jwhui 
I have reinstalled latest arm toolchain and now I am able to compile it.

closed 2018-11-15 by jwhui