openthread/openthread #3311: How to get network topology tool of thread?

tarunsmalviya, 2018-11-16:
I tried to built a tool to evaluate network topology using 'sudo wpanctl getprop'. But in this command, only the nodes which are directly connected to NCP are listed.

Can any help?

abtink, 2018-11-16:
There are multiple properties provided by OT/`wpantund` related to topology:
- `"Thread:ChildTable"` list of the children attached to the device
- `"Thread:NeighborTable"` list of neighbors (routers and direct chidllren)
- `"Thread:RouterTable"` list of all routers in networks (this list includes both routers that device can directly talk to, and routers that hops aways and the best next hop to reach them).

tarunsmalviya, 2018-11-16:
Yes. But I cannot create complete graph of network based on these information because this doesn't provide details of child which are not directly connected to NCP.

abtink, 2018-11-16:
Each router tracks its own children, so need to get the date from each router node separately.

jwhui, 2018-11-16:
By design, no Thread device needs to maintain state about all other Thread devices in the network - allowing the network to scale while minimizing memory and communication resource utilization.

As @abtink mentioned, the information needs to be aggregated across the nodes in the network.

tarunsmalviya, 2018-11-16:
Okay. Got it.
Thank you!

tarunsmalviya, 2018-11-16:
Okay.There are tools provided by Nordic which is for windows and Linux x64, in which a CLI is connected and the tools draws the network topology. Can you elaborate how did they do it.

jwhui, 2018-11-16:
Nordic implemented an application layer protocol where nodes report topology information to the topolgy visualization service.

LuDuda, 2018-11-16:
@tarunsmalviya @jwhui Actually, I am the author of nRF Thread Topology Monitor and just to be precise the tool communicates with CLI device which periodically retrieves the current list of routers. Each router is polled by sending the standard Thread diagnostic messages (MGMT_DIAG.req) - [CLI command](https://github.com/openthread/openthread/tree/master/src/cli#networkdiagnostic-get-addr-type-). In response, routers in the network send their current link set (current connected neighbors), child table, and basic network parameters, such as Network Address or Extended MAC Address.

Because of that, no additional application layer protocol is needed - just bare Thread stack.

It would be of course nice to add support for MGMT_DIAG messages to spinel so that apart of CLI, also NCP could be used as a topology monitor device :)

jwhui, 2018-11-16:
@LuDuda, thanks for clarifying! Adding support to spinel is an interesting thought. It might also be useful to implement a library so that any device that can route to the Thread network can perform these services.

LuDuda, 2018-11-16:
@jwhui Can you elaborate more on what type of library are you thinking of?

Currently, we use almost unmodified CLI example (we do have some fancy led blinking :)) that is entirely driven from PC side by node.js script (with logic for polling all routers, topology database, visualization etc).

jwhui, 2018-11-16:
@LuDuda use of TMF messages are limited to mesh-local addressing. It might also be nice to have something that does not require to be on-mesh. One possibility is to have something that operates in commissioner role and uses UDP proxy feature of border router. Anything more involved probably requires additional code on device and is probably best done within application-specific frameworks.

closed 2018-11-20 by jwhui

tarunsmalviya, 2018-11-20:
@LuDuda Can you please share more details regarding [https://github.com/openthread/openthread/tree/master/src/cli#networkdiagnostic-get-addr-type-](CLI) command as I didn't find any proper documentation regarding the CLI command.

Thanks

jwhui, 2018-11-20:
@tarunsmalviya , the [`networkdiagnostic get`](https://github.com/openthread/openthread/tree/cfc4882c3d7ee9f49ca9169394f61e7b43a56dd6/src/cli#networkdiagnostic-get-addr-type-) CLI command sends a `DIAG_GET.req` message to retrieve Thread state information from another device. You can learn more about available network diagnostics in [Thread 1.1.1 Section 10.11](https://www.threadgroup.org/support#specifications).

In summary, you can try the following steps:
1. Use the [`router table`](https://github.com/openthread/openthread/tree/cfc4882c3d7ee9f49ca9169394f61e7b43a56dd6/src/cli#router-table) CLI command to obtain the list of Thread routers in the network.
1. Use the [`networkdiagnostic get`](https://github.com/openthread/openthread/tree/cfc4882c3d7ee9f49ca9169394f61e7b43a56dd6/src/cli#networkdiagnostic-get-addr-type-) CLI command to retrieve information from a given Thread router.
1. You can see the information available via network diagnostics defined in [`src/core/thread/network_diagnostic_tlvs.hpp`](https://github.com/openthread/openthread/blob/cfc4882c3d7ee9f49ca9169394f61e7b43a56dd6/src/core/thread/network_diagnostic_tlvs.hpp).

Hope that helps.

tarunsmalviya, 2018-11-20:
@jwhui Thanks for the help :)

tarunsmalviya, 2019-03-18:
@LuDuda Can you share the complete format of command you mentioned. According to document, the commnd is `networkdiagnostic get <addr> <type>`. But what are values of <type> and what information does this command send. 
Also, please share any document regarding the same.

tarunsmalviya, 2019-03-19:
@jwhui `router table` command returns the list of router with details like rloc16, etc.

When running `networkdiagnostic get` command which address we have to pass as argument because I tried with passing rloc16, but got parse error.

Also, I am unable to fetch NCP data when I run `networkdiagnostic get` command by passing multicast address.

LuDuda, 2019-03-19:
@tarunsmalviya as described in the [documentation](https://github.com/openthread/openthread/tree/master/src/cli#networkdiagnostic-get-addr-type-) you need to pass RLOC IPv6 address, which is constructed from Mesh Local Prefix and RLOC16 - `MeshLocalPrefix::ff:fe00:RLOC16` e.g. `networkdiagnostic get fdde:ad00:beef:0:0:ff:fe00:f400 0 1 6`

tarunsmalviya, 2019-03-28:
@LuDuda How do we calculate connections between router using router table. According to me, links between router can be known using id and next hop field, but in output of router table command, invalid router id is there in next hop, i.e., 63. Does that mean something?

LuDuda, 2019-03-28:
In the TMM i just query for `Route64 `which helps to determine Routers topology. You can also query for `Child Table` to see attached children.

tarunsmalviya, 2019-03-28:
@LuDuda Thanks!
Can you please explain how to determine link between routers using Route64?

jwhui, 2019-03-28:
@tarunsmalviya, see [Thread 1.1.1](https://www.threadgroup.org/support#specifications) Section 5.20.1 for the MLE Route64 TLV Format.

Note that this issue was previously closed. Feel free to submit a new issue for future questions so that we can track them better. Thanks.