openthread/openthread #3324: Message size over 1280 Bytes

ChrIgiSta, 2018-11-21:
Is there a way to send messages (e.g.) with udp, which is over 1280 Bytes?
I'm not be able to send udp packages with more than 1232 Bytes ( +8 Bytes UDP Header + 40 Bytes Ipv6 Header = 1280 Bytes in total) from a Openthread Device to an other.
I try it on a NRF52 and on POSIX Simulation. The situation is in both case the same.

I increase for this case `OPENTHREAD_CONFIG_IPV6_DEFAULT_MAX_DATAGRAM` to 2*1280 Bytes.

Thanks for help.

jwhui, 2018-11-22:
Yes, increasing `OPENTHREAD_CONFIG_IPV6_DEFAULT_MAX_DATAGRAM` will allow you to send larger packets. Note, however, that the Fragmentation Header defined in [RFC 4944 Section 5.3](https://tools.ietf.org/html/rfc4944#section-5.3) only has 11 bits for the datagram length, which means that packets are limited to 2047 bytes.

abtink, 2018-11-27:
@ChrIgiSta using NCP mode, I think you also need to update the max spinel frame size. This is  defined in `spinel.h` header file `SPINEL_FRAME_MAX_SIZE,` and used only in `spinel.c` within OpenThread code.

Since `spinel.h` is intended to be shared between host and NCP code, we cannot use OT configuration header to change its definitions. 

If you are using `wpantund`, the spinel max size comes into picture again (in definitions of `mInboundFrame` and `mOutboundBuffer` and `mOutboundBufferEscaped` buffers in `SpinelNcpInstance`).

There may be other definitions in `wpantund` that impact MTU size. I know `TunnelIPv6Interface()` constructor has a `mtu` parameter with default value being 1280.

abtink, 2018-11-27:
BTW, we technically don't need `OPENTHREAD_CONFIG_NCP_TX_BUFFER_SIZE` to scale with the max IPv6 message size (_though increasing it would help with other spinel frames so I suggest to keep as large as possible_). 

The `NcpFrameBuffer` (when embedding an IPv6 `Message` into a spinel frame) does not copy the payload of `Message` into its buffer and instead just saves the `Message` pointer. It frees the `Message` instance once the content is read.

ChrIgiSta, 2018-11-29:
@abtink many thanks for answer.
I'm now be able to set the MTU of WPAN up to 2047 Bytes, and it works with the Border Router.

For people who also want (must) increase package size:
### NCP Device
 * `openthread-core-default-config.h` -> `OPENTHREAD_CONFIG_IPV6_DEFAULT_MAX_DATAGRAM` up to 2047   
 _and_ `OPENTHREAD_CONFIG_DEFAULT_SED_BUFFER_SIZE` up to 2047
 * `src/ncp/spinel.h` -> `SPINEL_FRAME_MAX_SIZE` ~2067
### wpantund @ borderrouter
 * `third_party/openthread/src/ncp/spinel.h` -> `SPINEL_FRAME_MAX_SIZE` ~2067
 * `src/util/TunnelIPv6Interface.h` -> `TunnelIPv6Interface(const std::string& interface_name = "", int mtu = 2047);` (sets the MTU from the WPAN0 Device)
### For Thread Devices
* `openthread-core-default-config.h` -> `OPENTHREAD_CONFIG_IPV6_DEFAULT_MAX_DATAGRAM` up to 2047

ChrIgiSta, 2018-11-29:
On last question:
Is there a reason, why openthread doesn't handle fragmented Ipv6 packets?
If a Ipv6 fragment is detected, the `HandleFragment` called in `ip6.cpp`. In this function a check `VerifyOrExit(fragmentHeader.GetOffset() == 0 && fragmentHeader.IsMoreFlagSet() == false, error = OT_ERROR_DROP);` drop every by ipv6 layer fragmented packets.

Thanks

jwhui, 2018-11-29:
@ChrIgiSta, thanks for listing the set of required changes!

To answer your question [Thread 1.1.1 Section 5.1](https://www.threadgroup.org/support#specifications) states that [RFC 2460 Section 4.5 Fragment Header](https://tools.ietf.org/html/rfc2460#section-4.5) is not required. However, I could see having IPv6 Fragment Header support as a configurable build option.

ChrIgiSta, 2018-12-03:
@jwhui, thanks for answer!

closed 2018-12-03 by ChrIgiSta