openthread/openthread #3325: OpenThread implementations for .Net Core

twinklekumarp, 2018-11-23:
Hi
As I am trying to find and follow OpenThread implementations for Microsoft .Net, I haven't found anything significant so far. It would be great if we able to implement OT for .Net core or .Net standard similar like OpenWeave so that other developers can take advantage of OT and language would not be barrier for them.

**Describe the solution you'd like**
Implementation of OT for .Net core or .Net standard as library and released as Nuget package.

**Describe alternatives you've considered**
At first I was thinking c++ but C# could be possible too.

**Additional context**
If provided chance I would like to contribute to .Net implementations. I have good experience with framework development in .Net and with your guidance and support I think I can do it.

jwhui, 2018-11-26:
@twinklekumarp, I'm not very familiar with .Net Core. Can you provide more detail on the use cases you'd like to support? Are you looking to support a host running .Net Core, OpenThread running in .Net Core framework, and communicating via a serial connection to an 802.15.4 SoC? What host platforms are you targeting?

jwhui, 2019-01-17:
@twinklekumarp, any updates?

jwhui, 2019-02-01:
Closing stale issue.

closed 2019-02-01 by jwhui