openthread/openthread #3326: [efr32] EFR32 flash data page number problem

swordinrock123, 2018-11-26:
Hi, I have a question about the flash page number of EFR32.

In my understanding, the flash is used for saving settings and config.
In [settings_flash.c](https://github.com/openthread/openthread/blob/master/examples/platforms/utils/settings_flash.c#L116), SETTINGS_CONFIG_PAGE_SIZE and SETTINGS_CONFIG_PAGE_NUM are used.

And for efr32, [SETTINGS_CONFIG_PAGE_SIZE](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/openthread-core-efr32-config.h#L93) is 2KB and [SETTINGS_CONFIG_PAGE_NUM](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/openthread-core-efr32-config.h#L101) is 4, so totally 8KB is required.

However, in [flash.c](https://github.com/openthread/openthread/blob/master/examples/platforms/efr32/flash.c#L43), the FLASH_DATA_USED_PAGE is 10, FLASH_PAGE_SIZE is 2KB, so totally 20KB is reserved.

I doubt whether this will cause flash memory waste issue, any thought?

jwhui, 2018-11-26:
Good catch. I believe you are correct. As you noted, it appears that the configuration in [examples/platforms/efr32/openthread-core-efr32-config.h](https://github.com/openthread/openthread/blob/0190f9bd9c3df6069b4d1740b9bf2dc8e77b700c/examples/platforms/efr32/openthread-core-efr32-config.h) uses fewer flash pages than [examples/platforms/efr32/flash.c](https://github.com/openthread/openthread/blob/0190f9bd9c3df6069b4d1740b9bf2dc8e77b700c/examples/platforms/efr32/flash.c) reserves. Ideally, the hard-coded values in [examples/platforms/efr32/flash.c](https://github.com/openthread/openthread/blob/0190f9bd9c3df6069b4d1740b9bf2dc8e77b700c/examples/platforms/efr32/flash.c) would be replaced with the configuration macros.

jwhui, 2019-02-22:
@swordinrock123, submitted #3620 to address this issue.

PR #3620 (2019-02-25): [efr32] unify flash config around SETTINGS_CONFIG_PAGE_NUM
Resolves #3326.
closed 2019-02-25 by jwhui