openthread/openthread #3346: [configure] plattforms not build without example

ChrIgiSta, 2018-12-04:
Since [commit `a638442df81981dd0ad8cce85c1d493d4c594441`](https://github.com/openthread/openthread/commit/a638442df81981dd0ad8cce85c1d493d4c594441#diff-51285cdd4615fad5900cc237c3cd412c), the directory `platforms` only included, if `OPENTHREAD_ENABLE_EXAMPLES` is `1`.
Before this commit was `platforms` always included by [`examples/Makefile.am`](https://github.com/openthread/openthread/commit/a638442df81981dd0ad8cce85c1d493d4c594441#diff-51285cdd4615fad5900cc237c3cd412c) and build. Is this change wanted, or only a mistake, happen by change `OPENTHREAD_EXAMPLES` to `OPENTHREAD_ENABLE_EXAMPLES`?
   
I ask this, because we want to use a newer openthread version by Zephyr, witch used  `platforms` without `OPENTHREAD_ENABLE_EXAMPLES`.

Thanks.

jwhui, 2018-12-04:
@ChrIgiSta, that was intentional at the time. Are you looking to have OpenThread build code in [examples/platforms/utils](https://github.com/openthread/openthread/tree/4c783e558422d27279ad0812899b1f641dec8643/examples/platforms/utils)? Everything else is platform-specific code and requires specifying a platform.

ChrIgiSta, 2018-12-05:
@jwhui, exactly. they use parts of [examples/platforms/utils](https://github.com/openthread/openthread/tree/4c783e558422d27279ad0812899b1f641dec8643/examples/platforms/utils).

jwhui, 2018-12-06:
@ChrIgiSta, makes sense. I submitted #3355 to resolve this issue. Please review.

ChrIgiSta, 2018-12-07:
@jwhui, i have reviewed the PR. Thanks for this change. :+1:

PR #3355 (2018-12-07): [examples] always build examples/platforms/utils
This PR resolves #3346.
closed 2018-12-07 by jwhui