openthread/openthread #3352: make fails for EFR32 Silicon Labs

Puchinger, 2018-12-05:
I made executed exactly the same commands as described here:
https://github.com/openthread/openthread/tree/master/examples/platforms/efr32
And get following error after executing make -f examples/Makefile-efr32 BOARD=BRD4161A
**To Reproduce**
Information to reproduce the behavior, including:
1. git clone https://github.com/openthread/openthread.git
$ cd <path-to-openthread>
$ ./script/bootstrap
$ cd <path-to-openthread>/third_party
$ mkdir silabs
$ cd <path-to-Simplicity-Studio>/developer/sdks
$ cp -rf gecko_sdk_suite <path-to-openthread>/third_party/silabs/
$ cd <path-to-openthread>
$ ./bootstrap
$ make -f examples/Makefile-efr32 BOARD=BRD4161A

....
compilation terminated.
In file included from openthread/examples/../third_party/mbedtls/mbedtls-config.h:168:0,
                 from openthread/examples/../third_party/mbedtls/repo/library/ccm.c:34:
openthread/examples/../examples/platforms/efr32/crypto/efr32-mbedtls-config.h:32:23: fatal error: em_device.h: No such file or directory

Is there an issue or doing am I something wrong?

Thanks in advance

jwhui, 2018-12-05:
What version of Flex SDK do you have installed?

The existing EFR32 port requires Flex SDK v2.3. I believe the latest Flex SDK is v2.4, so be sure to install Flex SDK v2.3 as well. After copying, you should have the following directory: `third_party/silabs/gecko_sdk_suite/v2.3`

Puchinger, 2018-12-06:
Tanks for your answer, using Flex SDK v2.3 solved the problem.
Maybe you could add the hint to use version 2.3 into the OpenThread on EFR32 Readme?

jwhui, 2018-12-06:
@Puchinger, we'll update the README. Thanks for raising this issue.