openthread/openthread #3362: otInstanceInit struct allocation

jia200x, 2018-12-10:
Hello OpenThread developers,

I'm currently improving the RIOT-OS port of OpenThread and came here to describe a problem I'm facing and a possible solution.

I'm trying to create an OpenThread container struct like this:
```
typedef struct {
```
Currently the OpenThread instances [are allocated in the core](https://github.com/openthread/openthread/blob/8d125135914e035eaf25270e5ed1e14061425e89/src/core/common/instance.cpp#L49).

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jia200x, 2018-12-10:
(sorry, it published before I finished. I'm still writing the description)

jia200x, 2018-12-10:
(done!)

jwhui, 2019-03-22:
@jia200x, OpenThread's [multiple instance feature](https://github.com/openthread/openthread/blob/23ff1015d4f68b07c303ab35c9656f43642d2c1d/configure.ac#L386) allows external allocation of OpenThread instance memory. Does that satisfy your use case?

jwhui, 2019-04-16:
@jia200x, any comments?

jia200x, 2019-04-17:
@jwhui sorry for the delay.

Yes, I think there could be a workaround with this approach for multiple instance feature. 
It would be direct though, if there's a way to know the size of the `otInstance` object in compile time (instead of calling `otInstanceInit(NULL, &len)`.) 

Do you know if this is possible?

jia200x, 2019-04-17:
well, I think my second question is unrelated to the original. So I guess the issue could be closed.
Thanks!

closed 2019-04-17 by jwhui