openthread/openthread #3363: [nrf52840] Elevated number of dropped packets and ChannelAccessFailures

avtolstoy, 2018-12-10:
**Describe the bug**
After upgrading OpenThread from f089fc8e25f6ce5bea9ef6c1bb11b234c4540c6a to 545ccd8dc8aeb360449cde43349f7ec633aa50e7 we are seeing elevated number of dropped packets and ChannelAccessFailures, which cause the connectivity between the nodes to be broken periodically.

**Console/log output**
The logs were collected from the border router node while performing an OTA upgrade of one of the nodes.
[545ccd8dc8aeb360449cde43349f7ec633aa50e7.log](https://github.com/openthread/openthread/files/2663779/545ccd8dc8aeb360449cde43349f7ec633aa50e7.log)
[f089fc8e25f6ce5bea9ef6c1bb11b234c4540c6a.log](https://github.com/openthread/openthread/files/2663780/f089fc8e25f6ce5bea9ef6c1bb11b234c4540c6a.log)

```
$ cat f089fc8e25f6ce5bea9ef6c1bb11b234c4540c6a.log | grep -i 'error' | grep -v 'DestinationAddressFiltered' | wc -l
69
$ cat 545ccd8dc8aeb360449cde43349f7ec633aa50e7.log | grep -i 'error' | grep -v 'DestinationAddressFiltered' | wc -l
385
```

hubertmis, 2018-12-17:
We've located the root cause of this problem. It is related to drift of high-frequency and low-frequency clocks. Recommended workaround is to change `NRF_RAAL_TIMESLOT_DEFAULT_MAX_LENGTH` value to `20000000`.
I'm working on proper fix.

tomas-c, 2018-12-18:
@avtolstoy What percentage of CCA errors were you seeing before and after the changes?  With EFR32 platform I can get CCA errors 10% of outgoing messages with 10 devices in one room. Wondering it that I should investigate that further too.

avtolstoy, 2018-12-19:
@tomas-c In a 4-node network after an OTA update (~700KB) of one of the nodes (through the border router):
- `f089fc8e25f6ce5bea9ef6c1bb11b234c4540c6a`: 0.9% dropped total, 0.6% due to `ChannelAccessFailure`
- `545ccd8dc8aeb360449cde43349f7ec633aa50e7`: 3.5% dropped total, 2.2% due to `ChannelAccessFailure`
- `545ccd8dc8aeb360449cde43349f7ec633aa50e7` with the workaround: 0.3% dropped total, 0.06% due to `ChannelAccessFailure`

But this is all circumstantial. As Hubert already mentioned, the issue is in nRF52840 radio driver and is caused by a drift between low-frequency and high-frequency clock.

jwhui, 2019-01-04:
Resolved by #3432.

closed 2019-01-04 by jwhui