openthread/openthread #3400: Ideal Poll period for SED

kpyush, 2018-12-24:
What would be the ideal poll period for the SED sending data every :
- 30 seconds
- 60 seconds
When I am increasing poll period from more than 2 seconds then SED number of reattach to parent increases and eventually average current increases.

jwhui, 2018-12-24:
@kpyush, are you using the [otThreadSetChildTimeout()](https://openthread.io/reference/group/api-thread-general#group__api-thread-general_1ga69319b9636a7f9b878fe663f64b0d835) API to change the parent's timeout value?

kpyush, 2018-12-25:
Yes I am using otThreadSetChildTimeout() API and current Timeout is 10 seconds. Is child timeout is causing the reattach?

jwhui, 2018-12-28:
As long as poll period is less than the child timeout, the SED should reliably maintain connection with its parent.

Can you provide more details on your setup? Including hardware platform, git commit id, build configuration, etc.

Also, can you enable logging and provide logs to help diagnose?

kpyush, 2018-12-31:
@jwhui problem has been resolved after increasing the child timeout > poll-period.

thanks

closed 2018-12-31 by kpyush