openthread/openthread #3442: DHCPv6 Client ID is carrying extended address instead of EUI64

kwangeom, 2019-01-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Information to reproduce the behavior, including:
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior**
A clear and concise description of what you expected to happen.

**Console/log output**
If applicable, add console/log output to help explain your problem.

**Additional context**
Add any other context about the problem here.

kwangeom, 2019-01-08:
I wanted dnsmasq from BBB to allocate fixed IPv6 address to the device based on EUI64 of the client device. But the client solict message is carrying extended address even though its duid hw type set to eui64 like below. And this extended address looks like randomly generated whenever reboot. Is there any reason to use extended address instead of eui64 here? If this is a bug, could you fix it?

otError Dhcp6Client::AppendClientIdentifier(Message &aMessage)
{
    ClientIdentifier option;

    option.Init();
    option.SetDuidType(kDuidLL);
    option.SetDuidHardwareType(kHardwareTypeEui64);
    option.SetDuidLinkLayerAddress(GetNetif().GetMac().GetExtAddress());
    return aMessage.Append(&option, sizeof(option));
}

Kwang

jwhui, 2019-01-08:
@kwangeom, thanks for raising this issue! I just submitted #3443 to resolve this issue. Please review.