openthread/openthread #3564: Parent RSSI values don't update

josephn123, 2019-02-07:
**Describe the bug**

Platform efr32. Build commit `f31f8c1f3c786a2465000e0ae18b85238618475f`, (latest, 6th Feb).

Parent RSSI value doesn't update, but corresponding entry in the neighbour table does. It seems the parent RSSI value is the value taken at the point or near the point of attaching to the parent, but not updated periodically afterwards.

**Steps to Reproduce**

- Attach two devices
- Disconnect the antennas
- Observe output from `wpanctl getprop`

**Output/Logs**

Output from `wpanctl getprop`. As you can see here, device `7E0F84C515403EF4` is the parent. In the `Thread:Parent` property it is listed with an Ave/Last RSSI of -39/-34dBm, whilst in the neighbour tables it is listed as -71/-72dBm.
```
Thread:NeighborTable = [
	"7E0F84C515403EF4, RLOC16:8c00, LQIn:3, AveRssi:-71, LastRssi:-72, Age:4, LinkFC:3989, MleFC:733, IsChild:no, RxOnIdle:yes, FTD:yes, SecDataReq:no, FullNetData:yes"
	"064C2EFFD9FCE9B7, RLOC16:d000, LQIn:3, AveRssi:-78, LastRssi:-81, Age:4, LinkFC:2270, MleFC:819, IsChild:no, RxOnIdle:yes, FTD:yes, SecDataReq:no, FullNetData:yes"
]
Thread:NeighborTable:ErrorRates = [
	"7E0F84C515403EF4, RLOC16:8c00, FrameErrRate:64.97%, MsgErrorRate:16.66%, AveRssi:-71, LastRssi:-72, "
	"064C2EFFD9FCE9B7, RLOC16:d000, FrameErrRate:25.23%, MsgErrorRate:5.95%, AveRssi:-78, LastRssi:-81, "
]
Thread:Parent = "7E0F84C515403EF4, RLOC16:8c00, Age:81, AveRssi:-39, LastRssi:-34, LQIn:3, LQOut:3"
```
**Expected Output**
`Thread:Parent` RSSI properties should read the same as the corresponding entries in the neighbour table.

jwhui, 2019-02-07:
@josephn123, I presume this device is a router-capable device. If so, this is expected behavior for the current implementation.

After attaching as a child, the device will eventually transition to a router role. In doing so, the device no longer maintains its parent information (a router has no parent), but instead maintains neighbor tables (the combination of neighboring routers and children).

While it might make sense to remove the parent information, it is currently necessary to support Thread Certification. The Thread Test Harness requires access to historical parent information to validate conformance of the Thread specification.

Does that answer your question?

josephn123, 2019-02-07:
Yes, makes sense, thank you

closed 2019-02-07 by josephn123