openthread/openthread #3565: [efr32] otPlatRadioGetRssi() is blocking

josephn123, 2019-02-07:
**Describe the bug**
`otPlatRadioGetRssi()` is a blocking API call on the efr32. Should we not avoid blocking the main execution context?

If `otPlatRadioEnergyScan()` was implemented (it's essentially an asynchronous equivalent of `otPlatRadioGetRssi()`), would `otPlatRadioGetRssi()` even be needed?

jwhui, 2019-02-07:
@josephn123, `otPlatRadioGetRssi()` is generally only used to perform energy scans. Implementing `otPlatRadioEnergyScan()` should avoid the use of `otPlatRadioGetRssi()` when performing energy scans. However, the optional jam detector component currently also uses `otPlatRadioGetRssi()`. If you don't use the jam detector, then no need to worry about it.

Specifically for `otPlatRadioGetRssi(), ideally non-blocking is best. It really comes down to how long `otPlatRadioGetRssi()` can block and whether or not the application using OpenThread can handle that.

josephn123, 2019-02-07:
As an improvement, is there any reason that the jam detector couldn't be refactored to use `otPlatRadioEnergyScan()` instead?

jwhui, 2019-02-07:
@abtink, thoughts?

abtink, 2019-02-07:
Jam detector module uses the MAC layer `EnergyScan()` API with `scan_duration` set to zero. 
What we expect from "zero-duration" scan is to take a single RSSI sample per channel in the scan mask. 
This is currently implemented in `Mac` using the `otPlatRadioGetRssi()` radio API.

Main reason for this is that to ensure the jam-detection sampling is done quickly and radio is not blocked for other operation for too long. In `otPlatRadioEnergyScan()` API, the scan time is in millisecond. So min we can use is 1ms (we want to avoid blocking radio for 16ms (1ms per channel) for jam detection).

The `GetRssi()` API should not take as long. From what I remember 15.4 spec defines the sample time for energy detection (i.e., getting a single RSSI) "The ED measurement time, to average over, shall be equal to 8 symbol periods." and with symbol rate of 62.5KSym/sec we get total of `128 usec` 

Also the  `GetRssi()` is expected to return the most recent RSSI measured value. So if the radio is continuously performing ED measurements and storing the value (and importantly channel has not changed recently), the latest measured value can be returned immediately. 

That said, if `otPlatRadioEnergyScan()` itself provides a zero-scan duration energy scan functionality (i.e., takes a single RSSI sample on given channel), the code can be changed to adopt that.

jwhui, 2019-04-16:
@josephn123, any further comments?

closed 2019-04-16 by josephn123