openthread/openthread #3582: Unable to communicate with NCP over ALOC

Nopik, 2019-02-11:
So, this case is pretty convoluted and involves some non-trivial use cases. I'll try make it clear:

a) in our software we do use Service ALOC to talk from any node within mesh to nearest Border Router. BR is then passing such request, e.g. through MQTT to some external service (in reality that is much more complex, but the minimal case is as described here)

b) we do want 2-way coap communication between NCP and linux host

c) node joining flow with external commissioner should work

The summary of the problem is: the 3 above requirements seem to be exclusive.

I'll start detailed description with presentation of the actors:

- If `SetReceiveIp6FilterEnabled( false )` is called, some scenarios do work, some not
- In `wpantund.conf` there are `Thread:Config:FilterRLOCAddresses` and `Thread:Config:FilterALOCAddresses` options, which do influence the result as well.

So far it seems that we can tweak only those 2 settings, with some settings a) does work, but b) doesn't, with some settings it is the other way around. Sometimes a) and b) are OK, but c) isn't.

Originally we played only with `FilterRLOCAddresses`, but it turns out that leaving it on default (`true`) and setting `FilterALOCAddresses` to `false` is better way.

If we don't filter ALOC addresses and do call `SetReceiveIp6FilterEnabled( false )` on NCP side, a) does work. However, in such scenario some packets (specifically those packets, which are sent to external Commissioner during node joining process) are delivered twice, and Commissioner's DTLS session gets crazy. This is something we didn't analyzed yet (why some packets are duplicated), and we don't understand it fully yet. But since it happens after we disable some filters, it seems to be a consequence.

Enabling ip6 filter (i.e. not calling the abovementioned function) causes joining flow to work fine, though ALOC communication (i.e. `a)` case) stops working. So far we've checked that relaxing `https://github.com/openthread/openthread/blob/master/src/core/net/ip6.cpp#L705` a bit, to pass Service ALOCs seems to do the trick.

Do we miss something? Or, is there a better way to fix the code than patching `Ip6::ProcessReceiveCallback`?

jwhui, 2019-02-12:
@Nopik, thanks for raising this issue!

You are running into the unfortunate uses of RLOC and ALOC to address services that are implemented on both the NCP and host. The fix you proposed in #3585 makes a lot of sense, since I expect Service ALOC endpoints to be implemented on the host.

Longer term, you may want to consider migrating to an architecture where OpenThread core is running on the host (see [OpenThread POSIX app](https://github.com/openthread/openthread/tree/038edf53b2d007fc5f31b22e2775e9c9579c2c9f/src/posix)). The POSIX app also includes a "Platform UDP" option, which puts most OpenThread services on the host's UDP socket/network stack, which avoids this awkward splitting of traffic based on IPv6 address.

Nopik, 2019-02-13:
Great, thanks for clarification. We will investigate the host UDP approach (`ot-ncp-radio` version, right?), as it indeed seems very promising.

closed 2019-02-13 by Nopik