openthread/openthread #3599: how to send and receive ipv6 packet throw thread protocol with nrf52840

testter2, 2019-02-17:
I'm working on a project with nrf52840 and i'm trying to send and receive ipv6 packet throw thread protocol so i tried to use to send with otIp6Send and receive with otIp6SetReceiveCallback but code doesn't see the function that i pass to otIp6SetReceiveCallback so i don't can't receive from transmitter.

i also tried to handle testing example(packet_error_rate) in nRF5SDKforThreadv 0.11 sdk which sends udp packets  so i can remove udp layer  and work with ip6 functions of openthread but i stuck in the same problem.

 
questions :
are there any clear steps to send and receive ip6 packets over thread and what needs to be initialized ?
any clear example that sends and receive ip6 packets over thread ?

jwhui, 2019-02-18:
Can you clarify what you mean by: "code doesn't see the function that i pass to otIp6SetReceiveCallback"?

testter2, 2019-02-19:
i'm working on thread_simple_coap example from nordic with some changes
in server side : we added case for button 2 to set new receiving callback function when receiving  an ipv6 packet. and at this function we print received packet.
in client side : we added case for button 2 to send ipv6 with otIp6Send function after we getting addresses from provisioning_response_handler function and construct header as mentioned before

main at server side (receiver)

```
/**
 * Copyright (c) 2017 - 2017, Nordic Semiconductor ASA
 * 
 * All rights reserved.
 * 
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 * 
 * 1. Redistributions of source code must retain the above copyright notice, this
 *    list of conditions and the following disclaimer.
 * 
 * 2. Redistributions in binary form, except as embedded into a Nordic
 *    Semiconductor ASA integrated circuit in a product or a software update for
 *    such product, must reproduce the above copyright notice, this list of
 *    conditions and the following disclaimer in the documentation and/or other
 *    materials provided with the distribution.
 * 
 * 3. Neither the name of Nordic Semiconductor ASA nor the names of its
 *    contributors may be used to endorse or promote products derived from this
 *    software without specific prior written permission.
 * 
 * 4. This software, with or without modification, must only be used with a
 *    Nordic Semiconductor ASA integrated circuit.
 * 
 * 5. Any software provided in binary form under this license must not be reverse
 *    engineered, decompiled, modified and/or disassembled.
 * 
 * THIS SOFTWARE IS PROVIDED BY NORDIC SEMICONDUCTOR ASA "AS IS" AND ANY EXPRESS
 * OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES
 * OF MERCHANTABILITY, NONINFRINGEMENT, AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL NORDIC SEMICONDUCTOR ASA OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
 * GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT
 * OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 * 
 */
/** @file
 *
 * @defgroup simple_coap_server_example_main main.c
 * @{
 * @ingroup simple_coap_server_example_example
 * @brief Simple CoAP Server Example Application main file.
 *
 * @details This example demonstrates a CoAP server application that provides resources to control BSP_LED_3
 *          via CoAP messages. It can be controlled by a board with related Simple CoAP Server application.
 */

#include <stdint.h>

#include "app_timer.h"
#include "bsp_thread.h"
#include "nrf_log_ctrl.h"
#include "nrf_log.h"
#include "nrf_log_default_backends.h"

#include "thread_coap_utils.h"
#include "thread_utils.h"

#include <openthread/openthread.h>

[... 139 lines cut ...]
    // Initialize Thread CoAP utils.
    thread_coap_utils_led_timer_init();
    thread_coap_utils_provisioning_timer_init();
    thread_coap_utils_led_timer_start();

    // Initialize the Thread stack.
    thread_instance_init();
    thread_coap_init();
    thread_bsp_init();

    while (true)
    {
        thread_process();
    if (NRF_LOG_PROCESS() == false)
        {
            thread_sleep();

        }
    }
}

/**
 *@}
 **/
```

jwhui, 2019-02-23:
I believe this is due to your IPv6 packet not being formatted properly at the sender, as mentioned in #3600.

testter2, 2019-02-23:
> 
> 
> I believe this is due to your IPv6 packet not being formatted properly at the sender, as mentioned in #3600.

this the address i'm using now
header[2]=(0xADDEFD);
header[3]=(0xEFBE);
header[4]=(0xFF000000);
header[5]=(0x6400FE);

and that's the addresses i'm getting from ipaddr command
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:6400 (i'm using that address)
fe80:0:0:0:e036:1ed7:3316:76dd
fdde:ad00:beef:0:660b:9537:5373:7a2e

testter2, 2019-02-23:
this's the packet i'm trying to send 
start of header as : version,traffic class,flow label,payload length,next header,hop limit
source address : fdde:ad00:beef:0:0:ff:fe00:6400
destination address : FF030000 : : 1
data : aData [20]={1,0,2,3,1,0,2,3,1,0,2,3,1,0,2,3,1,0,2,3};

jwhui, 2019-03-04:
Duplicate of #3600.

closed 2019-03-04 by jwhui