openthread/openthread #3600: otIp6Send doesn't work

testter2, 2019-02-18:
i'm trying to send ipv6 packet over thread with function otIp6Send.
but otIp6Send doesn't work and return '6' which means  "Failed to parse message or arguments." as openthread errors says.
i don't know why !!

i'm using nrf52840.
this's part of my code
 uint32_t* contructip6add (uint32_t* src,uint32_t* dest,uint16_t pay_len)
{
   static uint32_t header [10];
    header[0]=(((uint32_t)ver)<<28)|(((uint32_t)TRAFFIC_CLASS)<<20)|FLOW_LABEL;
    header[1]=(((uint32_t)pay_len)<<16)|(((uint32_t)NEXT_HEADER)<<8)|HOP_LIMIT;
    header[2]=src[0];
    header[3]=src[1];
    header[4]=src[2];
    header[5]=src[3]; 
    header[6]=dest[0];
    header[7]=dest[1];
    header[8]=dest[2];
    header[9]=dest[3];
    return header;
}

static void ip_packet_transmit(void)
{
      uint8_t err;
      uint8_t* header5;
      header5= (uint8_t*)contructip6add(source_addr,des_addr,4);
          uint8_t aData [4]={'p','l','a','y'};
        otMessage *amessage;
        amessage=otIp6NewMessage(thread_ot_instance_get(), false); 
                err=otMessageAppend(amessage, header5, 40);
        if (amessage == NULL) {
		 return 0;
	}  
        err=otMessageAppend(amessage, aData, 4);
             if (err!= OT_ERROR_NONE ) {
		return  0;
	}

        err=otIp6Send(thread_ot_instance_get(),amessage);
       if(err!= OT_ERROR_NONE)
       {
          NRF_LOG_INFO("fail to send with error =  %d",err);
       }
       else
      {
         NRF_LOG_INFO("sent");
       }
  
}

jwhui, 2019-02-18:
`OT_ERROR_PARSE` typically indicates that the IPv6 headers are not formatted properly.

Note that IPv6 uses big-endian byte ordering. However, your code does not seem to format the IPv6 header using big-endian byte ordering. Typically, you should use something like [`htons()`](https://linux.die.net/man/3/htons). You could also take a look at [src/core/net/ip6.hpp](https://github.com/openthread/openthread/blob/b0110db4e5d6578e6b104ef40080e9782a1a68de/src/core/net/ip6.hpp) to see how OpenThread formats IPv6 headers.

To help debug, I suggest you print out the raw bytes of your formatted IPv6 message and compare that to example IPv6 packets that you've captured (e.g. by using [`tcpdump`](https://www.tcpdump.org/)).

testter2, 2019-02-18:
> suggest you print out the raw bytes of your formatted IPv6 message and compare that to example IPv6 packets that you've captured

now using htonl
```
header[0]=htonl((((uint32_t)ver)<<28)|(((uint32_t)TRAFFIC_CLASS)<<20)|FLOW_LABEL);
header[1]=htonl((((uint32_t)pay_len)<<16)|(((uint32_t)NEXT_HEADER)<<8)|HOP_LIMIT);
header[2]=htonl(src[0]);
header[3]=htonl(src[1]);
header[4]=htonl(src[2]);
header[5]=htonl(src[3]); 
header[6]=htonl(dest[0]);
header[7]=htonl(dest[1]);
header[8]=htonl(dest[2]);
header[9]=htonl(dest[3]);
```

i have another problem as otIp6Send returns 2 which means "message was dropped." as openthread errors says.

testter2, 2019-02-18:
> 
> 
> > `OT_ERROR_PARSE` typically indicates that the IPv6 headers are not formatted properly.
> > Note that IPv6 uses big-endian byte ordering. However, your code does not seem to format the IPv6 header using big-endian byte ordering. Typically, you should use something like [`htons()`](https://linux.die.net/man/3/htons). You could also take a look at [src/core/net/ip6.hpp](https://github.com/openthread/openthread/blob/b0110db4e5d6578e6b104ef40080e9782a1a68de/src/core/net/ip6.hpp) to see how OpenThread formats IPv6 headers.
> > To help debug, I suggest you print out the raw bytes of your formatted IPv6 message and compare that to example IPv6 packets that you've captured (e.g. by using [`tcpdump`](https://www.tcpdump.org/)).
> 
> now using htonl
> 
> ```
>     header[0]=htons((((uint32_t)ver)<<28)|(((uint32_t)TRAFFIC_CLASS)<<20)|FLOW_LABEL);
>     header[1]=htons((((uint32_t)pay_len)<<16)|(((uint32_t)NEXT_HEADER)<<8)|HOP_LIMIT);
>     header[2]=htons(src[0]);
>     header[3]=htons(src[1]);
>     header[4]=htons(src[2]);
>     header[5]=htons(src[3]); 
>     header[6]=htons(dest[0]);
>     header[7]=htons(dest[1]);
>     header[8]=htons(dest[2]);
>     header[9]=htons(dest[3]);
> ```
> 
> i have another problem as otIp6Send returns 2 which means "message was dropped." as openthread errors says.

the problem was that header constructed in wrong way.

but now after editing the code
    header[0]=lwip_htonl((((uint32_t)ver)<<28)|(((uint32_t)TRAFFIC_CLASS)<<20)|FLOW_LABEL);
    header[1]=lwip_htonl((((uint32_t)pay_len)<<16)|(((uint32_t)NEXT_HEADER)<<8)|HOP_LIMIT);
    header[2]=(src[0]);
    header[3]=(src[1]);
    header[4]=(src[2]);
    header[5]=(src[3]);
    header[6]=(dest[0]);
    header[7]=(dest[1]);
    header[8]=(dest[2]);
    header[9]=(dest[3]);

i got no error from function otIp6Send . but i still can't receive the data from receiver as the receiver code doesn't get into my callback receiver function although when sending over coap ,receiver code get into my callback receive function without any problems.

jwhui, 2019-02-19:
Can you provide a hex dump of the IPv6 packet that you are trying to send as well as the coap message that is working for you?

testter2, 2019-02-23:
> 
> 
> Can you provide a hex dump of the IPv6 packet that you are trying to send as well as the coap message that is working for you?

this's the main of client example 
```

/**
 * Copyright (c) 2017 - 2017, Nordic Semiconductor ASA
 * 
 * All rights reserved.
 * 
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 * 
 * 1. Redistributions of source code must retain the above copyright notice, this
 *    list of conditions and the following disclaimer.
 * 
 * 2. Redistributions in binary form, except as embedded into a Nordic
 *    Semiconductor ASA integrated circuit in a product or a software update for
 *    such product, must reproduce the above copyright notice, this list of
 *    conditions and the following disclaimer in the documentation and/or other
 *    materials provided with the distribution.
 * 
 * 3. Neither the name of Nordic Semiconductor ASA nor the names of its
 *    contributors may be used to endorse or promote products derived from this
 *    software without specific prior written permission.
 * 
 * 4. This software, with or without modification, must only be used with a
 *    Nordic Semiconductor ASA integrated circuit.
 * 
 * 5. Any software provided in binary form under this license must not be reverse
 *    engineered, decompiled, modified and/or disassembled.
 * 
 * THIS SOFTWARE IS PROVIDED BY NORDIC SEMICONDUCTOR ASA "AS IS" AND ANY EXPRESS
 * OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES
 * OF MERCHANTABILITY, NONINFRINGEMENT, AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL NORDIC SEMICONDUCTOR ASA OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
 * GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT
 * OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 * 
 */
/** @file
 *
 * @defgroup simple_coap_client_example_main main.c
 * @{
 * @ingroup simple_coap_client_example_example
 * @brief Simple CoAP Client Example Application main file.
 *
 * @details This example demonstrates a CoAP client application that enables to control BSP_LED_0
 *          on a board with related Simple CoAP Server application via CoAP messages.
 *
 */

#include "app_timer.h"
#include "bsp_thread.h"
#include "nrf_log_ctrl.h"
#include "nrf_log.h"
#include "nrf_log_default_backends.h"

#include "thread_coap_utils.h"
#include "thread_utils.h"
#include <openthread/openthread.h>

#define IPV6_ADDR_BYTE_LENGTH   16                /**< Length of IPv6 address in bytes. */
#define PAYLOAD_BYTE_LENGTH     4                 /**< Length of test packet payload in bytes. */
[... 260 lines cut ...]
        bsp_board_led_on(0);
    log_init();
    timer_init();

    thread_instance_init();
    thread_coap_init();
    thread_bsp_init();
    

    while (true)
    {
        thread_process();

        if (NRF_LOG_PROCESS() == false)
        {
            thread_sleep();
        }
    }
}

/**
 *@}
 **/

```

jwhui, 2019-02-23:
Are you sure you have the IPv6 address correct? What you show above is:
```
    header[2]=(0xADDEFD);
    header[3]=(0xEFBE2);
    header[4]=(0xFF000000);
    header[5]=(0x00FC0FE);
```
On a little-endian machine, that translates to: `fdde:ad00:e2fb:0e00:0000:00ff:fec0:0f00`, which does not appear valid.

As I requested in a previous comment, can you provide a hex dump of the IPv6 packet that you are trying to send as well as the coap message that is working for you? From that, it should be pretty easy to tell what the difference is.

testter2, 2019-02-23:
this the address i'm using now
    header[2]=(0xADDEFD);
    header[3]=(0xEFBE);
    header[4]=(0xFF000000);
    header[5]=(0x6400FE);

and that's the addresses i'm getting from ipaddr command
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:6400 (i'm using that address)
fe80:0:0:0:e036:1ed7:3316:76dd
fdde:ad00:beef:0:660b:9537:5373:7a2e

jwhui, 2019-02-23:
It also appears that you are trying to send the messages with link security disabled. Thread requires all data packets (with the exception of commissioning-related ones) to be communicated with link security enabled.

Try changing the second argument to `otIp6NewMessage()` to `true`. For example:
```c++
otIp6NewMessage(thread_ot_instance_get(), true);
```

testter2, 2019-02-26:
> It also appears that you are trying to send the messages with link security disabled. Thread requires all data packets (with the exception of commissioning-related ones) to be communicated with link [...]

thank you now i can send and receive data but just one last thing ,when i receive data,i find byte number 5 always = 0 and byte number 6 = payload length .
is that normal ? is there a minimum payload length that i shouldn't use less than that minimum or not ?

jwhui, 2019-02-26:
[RFC 8200 Section 3](https://tools.ietf.org/html/rfc8200#section-3) specifies the IPv6 Header format. The Payload Length occupies octets 5 and 6 and the field is big-endian. As a result, the Payload Length must be >= 256 before you will see octet 5 as non-zero.

testter2, 2019-02-26:
> 
> 
> [RFC 8200 Section 3](https://tools.ietf.org/html/rfc8200#section-3) specifies the IPv6 Header format. The Payload Length occupies octets 5 and 6 and the field is big-endian. As a result, the Payload Length must be >= 256 before you will see octet 5 as non-zero.

i'm not talking about payload length field in the header but the data after source and destination.
i find in data field that byte number 5 always = 0 and byte number 6 = payload length .

jwhui, 2019-02-26:
Your code sets the IPv6 Next Header field to 17 (i.e. UDP).

[RFC 768](https://tools.ietf.org/html/rfc768) specifies the UDP Header format. The UDP Length occupies octets 5 and 6 and the field is big-endian. As a result, the Payload Length must be >= 256 before you will see octet 5 as non-zero.

closed 2019-03-04 by jwhui

testter2, 2019-03-06:
> 
> 
> Your code sets the IPv6 Next Header field to 17 (i.e. UDP).
> 
> [RFC 768](https://tools.ietf.org/html/rfc768) specifies the UDP Header format. The UDP Length occupies octets 5 and 6 and the field is big-endian. As a result, the UDP Length must be >= 256 before you will see octet 5 as non-zero.

thanks for your time and effort.
it works now as i sets ipv6 next header to 0x59. i don't know if it should be 59 or 0x59 as 59 means next header is null and that what i need but it doesn't send if i set it to 59