openthread/openthread #3619: Steering Data is wrong

kwangeom, 2019-02-22:
**Describe the bug**
I tested with two devices to join a thread network.
First device was successful in commissioning and then tried commissioning for the second device.
But it does not start DTLS session because of wrong "Steering Data" from "Discovery Response".
From the below log captures, when second device tries commissioning, the steering data sent from NCP looks for the first device.

This is first device wireshark log when success
[image]

This is second device wireshark log when success
[image]

This is second device wireshark log when fail
[image]

**To Reproduce**
Information to reproduce the behavior, including:
Network looks like this : BBB--NCP(leader)-----CLI MTD(1)
                                                                       |-- CLI MTD(2)
commission first device, and then commission the second device.
try this several times. sometimes second device does not start DTLS session after receiving Discovery Response. Then look into the streering data to see if it is ok.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Console/log output**
If applicable, add console/log output to help explain your problem.

**Additional context**
Add any other context about the problem here.

jwhui, 2019-02-22:
@kwangeom, thanks for raising this issue. Can you provide more detail on the exact commands you used to reproduce the behavior? For example, the `wpanctl` commands on the border router and `cli` commands on the joiner, and their order?

kwangeom, 2019-02-22:
from BBB side, I ran for the 2 devices,
root@beaglebone:/home/debian/borderrouter/tests/meshcop#  ./otbr-commissioner --                                                      network-name OpenThreadEom --xpanid 1111111122222222 --agent-passphrase 123456 -                                                      -joiner-eui64 00124b0014f41d01 --joiner-passphrase DRRLCK1 --agent-addr 192.168.                                                      2.2 --agent-port 49191 --comm-envelope-timeout 200 --commission-device --disable                                                      -syslog
./otbr-commissioner --network-name OpenThreadEom --xpanid 11111111222222                                                      22 --agent-passphrase 123456 --joiner-eui64 00124b0014f4356c --joiner-passphrase                                                       DRRLCK1 --agent-addr 192.168.2.2 --agent-port 49191 --comm-envelope-timeout 200                                                       --commission-device --disable-syslog

from cli side, I ran "joiner start DRRLCK1".

jwhui, 2019-02-22:
Can you also increase the debug level (using the `-d` option) and capture the log output for the commissioner?

jwhui, 2019-02-22:
@kwangeom, after looking at this a bit, it appears that `otbr-commissioner` is not able to establish a DTLS session with the border agent until the previous session has timed out (~50 seconds).

@bukepo @gjc13, can you help take a look?