openthread/openthread #3635: [travis] posix-app-ncp test fails frequently

jwhui, 2019-02-27:
@bukepo, any thoughts?

bukepo, 2019-02-27:
Yes, this requires a more complicated implementation to ensure the synchronization of all components(`ot-ncp-radio`, `ot-ncp-?td`, `pyspinel-cli.py` and `node_cli.py`). I will look into this ASAP.

jwhui, 2019-03-12:
It appears Cert_7_1_03 is the one that fails most frequently and fails because the SED does not autoconfigure IPv6 addresses with the on-mesh prefix.

bukepo, 2019-03-12:
Maybe should temporarily disable the test case. I don't have time to fix it for now.

jwhui, 2019-03-12:
Just submitted #3683 to temporarily remove posix-app-ncp test.

LuDuda, 2019-03-12:
@bukepo from your first answer I see that you are aware of the root cause of this problem. Can you elaborate more on that? Is that issue only related to virtual time, or fundamentally RCP architecture is not stable?

bukepo, 2019-03-12:
Yes, the root cause was that when Travis is busy, the process scheduler may not schedule running order of processes of ot-ncp-radio, ot-ncp, pyspinel-cli and virtual time simulator as we expect. I added more virtual events to guarantee the order. However, I happened to think the order is not guaranteed in real situations, so I didn't submit the complicated version. I will look into this issue when I have time.

PR #4337 (2019-11-18): [travis] enable posix-app-ncp test
The `posix-app-ncp` test seems to be stable again.

Resolves #3635
closed 2019-11-18 by jwhui